tiny-pilot/tinypilot #37: Add support for OS Key doubleclick

mtlynch, 2020-06-25:
Sometimes the OS key is useful to press by itself regardless of key combinations. Allow double click to just force a modifier key event on its own.

somik123, 2020-07-24:
> Sometimes the OS key is useful to press by itself regardless of key combinations. Allow double click to just force a modifier key event on its own.

Some VM hosts monitor right cotnrol key as a modifier. If pressed, any key pressed after will be sent directly. Others like noVNC gives you a list of keys to press:

[image]

somik123, 2020-08-01:
@mtlynch most key combinations other then predefined hotkeys works fine. In my case, Alt+V brings up my clipboard manager and it is not passed to browser, thus not to tinypilot. Similarly, unable to send Win key (like you said on your first post).

PR #290 (2020-10-12): Support sending modifiers immediately with double click
Previously, the buttons in the web UI allowed the user to simulate holding a modifier key on the next keypress, but there was no way to send, for example, the OS keypress on its own. This change allows users to double-click the modifier key buttons to simulate a keypress of that particular key.

Fixes #37
commit fc7cbbc5 (2020-10-12): Support sending modifiers immediately with double click (#290)

Previously, the buttons in the web UI allowed the user to simulate holding a modifier key on the next keypress, but there was no way to send, for example, the OS keypress on its own. This change allows users to double-click the modifier key buttons to simulate a keypress of that particular key.

Fixes #37
closed 2020-10-12 by mtlynch