tiny-pilot/tinypilot #40: Support mouse input

mtlynch, 2020-06-26:
Mouse input capturing is doable. We can use a really similar mechanism to what we're already doing to emulate a keyboard:

https://www.raspberrypi.org/forums/viewtopic.php?t=234495

somik123, 2020-07-24:
> Mouse input capturing is doable. We can use a really similar mechanism to what we're already doing to emulate a keyboard:
> 
> https://www.raspberrypi.org/forums/viewtopic.php?t=234495

+1 on this.

Really need mouse support to finish installing a certain OS developed by Microsoft...

Lotusshaney, 2020-07-27:
This would be fantastic !

thetanil, 2020-07-28:
RE: Windows support
It seems that bitlocker (ms drive encryption) boots into some video mode which is not captured by the recommended (non-branded) HDMI capture device (or lost in the encoding). Should I open another issue for that?

Additionally the bitlocker screen times out and powers down about the time that the pi actually boots through to UI available. Just wanted to document somewhere that the bitlocker timeout can be disabled with the following command:

    bcdedit /set {bootmgr} bootshutdowndisabled 1

mtlynch, 2020-07-28:
@thetanil a separate issue would be great! Can you include the Windows version where you're seeing this?

somik123, 2020-08-01:
Any plans or timeline on this? Most installers support mouse so it'll make navigation easier.

mtlynch, 2020-08-01:
@somik123 - It's the top feature on my list. I'm aiming to have it working this month. I just need to find some bandwidth for it because a lot of my focus the past week has been helping new users get up and running and shipping out kits.

Bramzor, 2020-08-07:
+1

I need the same functionality in the coming weeks. (Do not need keyboard, only the mouse input). Maybe we can create a new branch and work on this together to get the basic functionality working?

peterberbec, 2020-08-11:
I, for one, welcome our new TinyPilot overlords.

ImCoKeMaN, 2020-08-16:
Would it be using this? https://developer.mozilla.org/en-US/docs/Web/API/Element/mousemove_event and then if you were to save the previous mouse position, capture new offset and do a diff, that's what the mouse gadget needs as input right?
If you're already past this stage, awesome, but this idea was in my head and wanted to share it if is on the right track.

mtlynch, 2020-08-17:
@ImCoKeMaN - Yep, that's my plan. Thanks for the pointer!

closed 2020-08-27 by mtlynch

sstur, 2020-08-29:
Hey, let me know if you need any support to help write the JS or any web frontend dev for this, I'll be happy to pitch in!

mtlynch, 2020-08-29:
@sstur - Thanks for reaching out! Is there a feature or piece you're interested in taking on? I think contributions work best when you can work on something you see as improving your experience using TinyPilot.

sstur, 2020-08-29:
Right, I was specifically thinking about the mouse support, just if you need anything on the JavaScript UI side to move that feature forward (although I'm happy to tackle any frontend feature). I'm still waiting on my parts that I ordered as soon as I discovered this project, so I'll have a better idea soon of what I'm interested in improving.

mtlynch, 2020-08-29:
The only big missing feature from mouse behavior I can think of at the moment is scrollwheel movement. That's probably going to be more backend-heavy, but it does involve a bit of frontend.

commit 6852dc25 (2020-08-18): Add support for mouse input

Fixes #40
commit 8b1e1d17 (2020-08-18): Add support for mouse input

Fixes #40
commit b67c7dd5 (2020-08-24): Add support for mouse input

Fixes #40
commit a4fbd3af (2020-08-25): Add support for mouse input

Fixes #40