tiny-pilot/tinypilot #64: Nginx "Bad gateway" after restarting TinyPilot

mtlynch, 2020-07-20:
There seems to be an issue where nginx will report "

### Repro steps

1. Start a TinyPilot device normally
1. `sudo service tinypilot restart`
1. Visit <http://tinypilot/>

### Expected

TinyPilot web interface loads normally.

### Actual

Nginx reports "Bad Gateway"

### Workaround

Restart nginx:

```bash
sudo service nginx restart
```

mtlynch, 2020-07-24:
From further testing, it seems that nginx is actually just "remembering" that the TinyPilot backend is down for a few seconds too long. If you just give it a few seconds, it resumes forwarding without the need for a restart. Still, I'd like to get rid of this behavior because if the backend is serving, nginx shouldn't be claiming bad gateway.

somik123, 2020-07-24:
> From further testing, it seems that nginx is actually just "remembering" that the TinyPilot backend is down for a few seconds too long. If you just give it a few seconds, it resumes forwarding [...]

What does your nginx configuration look like? I didn't seem to find it...

mtlynch, 2020-07-24:
Interesting, thanks @somik123! I'll play around with that and see if that improves things.

What version of nginx are you running? I'm running the version from the Debian repo, which is quite old:

```bash
nginx version: nginx/1.14.2
built with OpenSSL 1.1.1c  28 May 2019 (running with OpenSSL 1.1.1d  10 Sep 2019)
TLS SNI support enabled
configure arguments: --with-cc-opt='-g -O2 -fdebug-prefix-map=/build/nginx-7du1qr/nginx-1.14.2=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Wdate-time -D_FORTIFY_SOURCE=2' --with-ld-opt='-Wl,-z,relro -Wl,-z,now -fPIC' --prefix=/usr/share/nginx --conf-path=/etc/nginx/nginx.conf --http-log-path=/var/log/nginx/access.log --error-log-path=/var/log/nginx/error.log --lock-path=/var/lock/nginx.lock --pid-path=/run/nginx.pid --modules-path=/usr/lib/nginx/modules --http-client-body-temp-path=/var/lib/nginx/body --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --http-proxy-temp-path=/var/lib/nginx/proxy --http-scgi-temp-path=/var/lib/nginx/scgi --http-uwsgi-temp-path=/var/lib/nginx/uwsgi --with-debug --with-pcre-jit --with-http_ssl_module --with-http_stub_status_module --with-http_realip_module --with-http_auth_request_module --with-http_v2_module --with-http_dav_module --with-http_slice_module --with-threads --with-http_addition_module --with-http_geoip_module=dynamic --with-http_gunzip_module --with-http_gzip_static_module --with-http_image_filter_module=dynamic --with-http_sub_module --with-http_xslt_module=dynamic --with-stream=dynamic --with-stream_ssl_module --with-stream_ssl_preread_module --with-mail=dynamic --with-mail_ssl_module --add-dynamic-module=/build/nginx-7du1qr/nginx-1.14.2/debian/modules/http-auth-pam --add-dynamic-module=/build/nginx-7du1qr/nginx-1.14.2/debian/modules/http-dav-ext --add-dynamic-module=/build/nginx-7du1qr/nginx-1.14.2/debian/modules/http-echo --add-dynamic-module=/build/nginx-7du1qr/nginx-1.14.2/debian/modules/http-upstream-fair --add-dynamic-module=/build/nginx-7du1qr/nginx-1.14.2/debian/modules/http-subs-filter
```

I'm considering switching to a version that I compile from source so I can get a more recent stable version.

somik123, 2020-07-25:
@mtlynch I'm actually running a older version from ubuntu repo.

```
nginx version: nginx/1.14.0 (Ubuntu)
built with OpenSSL 1.1.1  11 Sep 2018
TLS SNI support enabled
configure arguments: --with-cc-opt='-g -O2 -fdebug-prefix-map=/build/nginx-GkiujU/nginx-1.14.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Wdate-time -D_FORTIFY_SOURCE=2' --with-ld-opt='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now -fPIC' --prefix=/usr/share/nginx --conf-path=/etc/nginx/nginx.conf --http-log-path=/var/log/nginx/access.log --error-log-path=/var/log/nginx/error.log --lock-path=/var/lock/nginx.lock --pid-path=/run/nginx.pid --modules-path=/usr/lib/nginx/modules --http-client-body-temp-path=/var/lib/nginx/body --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --http-proxy-temp-path=/var/lib/nginx/proxy --http-scgi-temp-path=/var/lib/nginx/scgi --http-uwsgi-temp-path=/var/lib/nginx/uwsgi --with-debug --with-pcre-jit --with-http_ssl_module --with-http_stub_status_module --with-http_realip_module --with-http_auth_request_module --with-http_v2_module --with-http_dav_module --with-http_slice_module --with-threads --with-http_addition_module --with-http_geoip_module=dynamic --with-http_gunzip_module --with-http_gzip_static_module --with-http_image_filter_module=dynamic --with-http_sub_module --with-http_xslt_module=dynamic --with-stream=dynamic --with-stream_ssl_module --with-mail=dynamic --with-mail_ssl_module
```

Bramzor, 2020-08-08:
"Bad gateway" is actually a valid answer during the reboot and shouldn't be confused with "Gateway Timeout".
A gateway timeout is that the service is running, but not providing an answer within the specified timeframe. Can be increased as mentioned by somik but this will not solve a "Bad gateway" issue.
A Bad gateway happens when Nginx tries to connect to the service but for some reason, that port is not up, blocked or tcp session could not be set up or any other reason.

So I assume Nginx is running before the service is ready? For this, there are a few possible ways to avoid this:

- (On service level) Start the nginx service after starting the end service by specifying an "After" in the .service file.
- (On nginx level) Using something like "health_check interval=10 fails=3 passes=1" in nginx which might help, haven't tried it myself yet. But this defines how long Nginx will see an origin as bad.
- (On application level) When clicking on reboot, use Ajax requests to check if the backend is available again and than refresh the browser.

mtlynch, 2020-08-08:
@Bramzor the issue I'm seeing is that TinyPilot resumes and responds to requests on port 8000 if I hit it directly, but if I hit port 80, nginx reports "Bad Gateway." This seems strange because the backend is back up, as far as I can tell. If I keep refreshing on port 80 for about 5 seconds, nginx resumes forwarding to TinyPilot, but I'd expect the forwarding to work as soon as TinyPilot is back up.

Bramzor, 2020-08-08:
@mtlynch Nginx keeps a status of the origin in cache. health_check for example has a default of "interval" of 5 seconds which means that if Nginx did a request to the origin and if it failed (default of "fails" is 1), it will cache the origin as bad for 5 seconds before it will retry the origin on a new request.

So you might want to set health_check interval to a value of 1s to lower the issue from 5s to 1s. Or increase "fails" to such a high number that the origin will never be specified as down.

mtlynch, 2020-08-09:
@Bramzor - oh, that's interesting. TinyPilot uses the free, open source version of nginx, though. The `health_check` directive is available only in the commercial version.

[image]

mtlynch, 2020-08-09:
Ohhh! I see. I can actually control this if I define explicit backends and use `fail_timeout` and `max_fails`.

Bramzor, 2020-08-09:
@mtlynch I think so yes. What do you see in the Nginx logs when you receive the error?

mtlynch, 2020-08-09:
I'd see lines like this:

```text
2020/08/08 20:57:14 [error] 12922#12922: *95 connect() failed (111: Connection refused) while connecting to upstream, client: 10.0.0.100, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "tinypilot"
```

But I confirmed a fix in https://github.com/mtlynch/ansible-role-tinypilot/pull/27

closed 2020-08-09 by mtlynch