tiny-pilot/tinypilot #75: Possibility to use g_mass_storage module?

ShaRose, 2020-07-23:
After digging into this a little bit (I don't have the hardware to test anything at the moment), it should be possible to use the g_mass_storage kernel module to also expose a block device or an ISO to the connected host.

Some limitations:

- It seems like that at the moment, without patches, it only supports acting as a CDROM, not DVD. Filesize limits are a problem.
- You might have to unload and reload the module to change the iso file path.

Could be interesting however.

mtlynch, 2020-07-23:
Yes, definitely possible. It's a feature I'm exploring.

jd-boyd, 2020-07-24:
If you support this, it might be nice to allow designating an external USB drive (stick or something on a SATA adapter) for storing the ISOs uploaded rather than writing them to the boot SSD card.

ShaRose, 2020-07-24:
I'd assume you'd be able to select anywhere you want on the filesystem: Or just hardcode a path for isos and you can mount whatever you want at that path, or just create links.

somik123, 2020-08-01:
+1 to this, though plugging in a flash drive works just as well. I mean you are going to plug in the Pi to the computers hdmi and usb anyway.

And from my experience, this sort of usb mass storage emulation suffer from really bad speeds. Dunno how well the Pi one will perform, but I can bet it wont be USB 2.0 speeds, let alone 3.0 speeds....

peterberbec, 2020-08-04:
I have personally spend over $300 for [this device](http://iodd.kr/wordpress/product/iodd-mini/), so this would be a huge value add. A virtual cdrom/dvd-rom that also is a kvm?

voltagex, 2021-05-09:
So from ServeTheHome's recent video it looks like this is supported, but it wasn't clear if this is a pro-only feature?

mtlynch, 2021-05-10:
Yes, this is unfortunately only available in the Pro version.

closed 2021-05-10 by mtlynch