tiny-pilot/tinypilot #76: VGA capture

steverweber, 2020-07-23:
These KVM devices are very helpful in loud server rooms when a system has no LOM.

Has anyone tried using a VGA adapter? will it work?
I ask because many rackmount servers only have VGA.

perhaps add a small blurb regarding VGA to the README.

Well done. Work for an university and we love opensource <3

lars18th, 2020-07-24:
Hi @steverweber ,

I don't tested it. However, I have one cheap VGA-2-HDMI converter. Perhaps someone can try this option.

The problem with _old_ KVMIP servers has the VGA analog capture. Today it's best to use cheap VGA capture to HDMI. These devices has chips that can handle standard VGA resolutions and convert them to HDMI signals. As we don't need broadcast grade videos, then it's fine if the image has sufficient quality. Perhaps not all resolutions are supported, and most can be windowed, but almost you get _clear_ images. Futhermore, the price of these devices is less than $10.

So, I suggest to try it:  SERVER --> VGA-2-HDMI CONVERTER --> HDMI USB 2.0 CAPTURE --> RASBERRY PI 4

I hope it helps.
Regards.

davidjericho, 2020-07-24:
I was literally doing this today. A SGI Octane's 13w3 sync-on-green connector, into HD15 VGA, passed to a VGA to HDMI convertor, into the HDMI capture. It was crisp and clear, and for the money paid I was very happy.

steverweber, 2020-07-24:
Looking on amazon to buy a VGA input to HDMI out but not having much luck. Perhaps i'm just not searching for the correct keywords. Could you pass a link? Thanks!

davidjericho, 2020-07-25:
https://www.amazon.com.au/gp/product/B07CTHPVB5/ is the exact product I used. I've since found the same thing on all the other usual online sites with alternate or no branding.

peterberbec, 2020-07-28:
[This adapter](https://www.amazon.com/dp/B07121Y1Z3) is confirmed as working great. TinyPilot - hdmi - adapter - computer
[image]

steverweber, 2020-07-28:
I also confirmed a cheap VGA to HDMI adapter works.
We can close this issue but first should include something in the README so other users are in the know.

mtlynch, 2020-07-28:
Awesome, thanks for confirming the working gear, @peterberbec! I updated the README in #95 to point to the device you're using.

closed 2020-07-28 by mtlynch

jorgehdz86, 2022-06-16:
hello, a few months ago I was using a vga converter and it worked without problem, a few days ago after an update it stopped working and does not show video. Even the DHMI works fine. I hope you can help me.

mtlynch, 2022-06-17:
@jorgehdz86 - Sorry to hear the video is cutting out! If you create a thread in our [help forum](https://forum.tinypilotkvm.com/), we can try to figure out what's going on.

PR #95 (2020-07-28): Document supported VGA to HDMI adapter
Fixes #76
commit e175e836 (2020-07-28): Document supported VGA to HDMI adapter (#95)

Fixes #76