tiny-pilot/tinypilot #79: Any ideas if this would work with an Odroid C2?

danmed, 2020-07-24:
I believe the Micro USB port on the Odroid C2 is OTG... so do you think it would work?

mtlynch, 2020-07-24:
It's possible. I don't have one on hand to check, but I'd welcome reports from anyone who does.

danmed, 2020-07-24:
> 
> 
> It's possible. I don't have one on hand to check, but I'd welcome reports from anyone who does.

I had a good go at it today.. Got debian stretch on an sd card and all updated. I had to make a couple of changes to the quick_install script.. Mainly either side of the ansible install.. Something like +u and -U above and below so that it would install..

But after that, it failed while building wheels.. something like "bdist wheel" as an invalid command...

I;ve not got it powered up at the moment so can't get exact errors.

mtlynch, 2020-07-24:
The bdist_wheel is an innocuous error as far as I can tell. It's a bug in Ansible's pip installation, but it seems to continue the install using another method. If it failed, there should be an error further down in the Ansible log.

danmed, 2020-07-24:
> 
> 
> The bdist_wheel is an innocuous error as far as I can tell. It's a bug in Ansible's pip installation, but it seems to continue the install using another method. If it failed, there should be an error further down in the Ansible log.

Ah, then maybe i shouldn't of quit the install :) I'll try it again tomorrow and get back to you.. hopefully i've ordered the correct hdmi to usb adapter..

https://www.amazon.co.uk/gp/product/B08C7B4DKM/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

I got this one because the one linked from your blog had a months wait time!

mtlynch, 2020-07-25:
That's a slightly different one, but all these products are just different packaging on top of the MacroSilicon MS2109 chip, so it should work exactly the same.

danmed, 2020-07-25:
> 
> 
> That's a slightly different one, but all these products are just different packaging on top of the same MacroSilicon MS2109 chip, so it should work exactly the same.

So i went back to a fresh debian install and ran through it again..

i had to change a small part of the script : 

`set +u
. venv/bin/activate
set -u
`
And this is how it finished : 

  `  running install
    running build
    running build_py
    creating build
    creating build/lib.linux-aarch64-3.5
    creating build/lib.linux-aarch64-3.5/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/verifier.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/error.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/pkgconfig.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/commontypes.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/cffi_opcode.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/backend_ctypes.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/recompiler.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/__init__.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/cparser.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/model.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/ffiplatform.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/api.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/lock.py -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/_embedding.h -> build/lib.linux-aarch64-3.5/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux-aarch64-3.5/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-aarch64-3.5
    creating build/temp.linux-aarch64-3.5/c
    aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-R9trRz/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/ffi -I/usr/include/libffi -I/tmp/tmp.3CkJZ2vPVE/venv/include -I/usr/include/python3.5m -c c/_cffi_backend.c -o build/temp.linux-aarch64-3.5/c/_cffi_backend.o
    unable to execute 'aarch64-linux-gnu-gcc': No such file or directory
    error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

    ----------------------------------------
Command "/tmp/tmp.3CkJZ2vPVE/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-3dg9wxx1/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-piwkymyq-record/install-record.txt --single-version-externally-managed --compile --install-headers /tmp/tmp.3CkJZ2vPVE/venv/include/site/python3.5/cffi" failed with error code 1 in /tmp/pip-build-3dg9wxx1/cffi/
`

mtlynch, 2020-07-25:
Oh, I forgot that ODroid is 64-bit. I haven't tested with aarch64 yet.

@KuroSaru [got TinyPilot working on an aarch64 beta](https://github.com/mtlynch/tinypilot/issues/83#issuecomment-663725908) of Raspberry Pi OS. They had to manually run `sudo apt-get install libffi-dev` but that doesn't look like it's your issue.

Would you be able to link to a gist of your full log? I can't tell from the snippet where the failure happened in the process.

danmed, 2020-07-25:
> Oh, I forgot that ODroid is 64-bit. I haven't tested with aarch64 yet.
>
> @KuroSaru [got TinyPilot working on an aarch64 beta](https://github.com/mtlynch/tinypilot/issues/83#issuecomment-663725908) of [...]

I would, but i seem to of corrupted the SD card again.. The power supply i'm using is a bit loose and it seems to have moved and shafted it.. if i get it going again i'll let you know... I may just buy a pi4 though to be honest.

danmed, 2020-07-25:
OK, so it's back up.. Here's the gist : 

https://gist.github.com/danmed/aeef59f4485c8c6c06a5d67e1ecf9817

mtlynch, 2020-07-25:
Thanks! It looks like it's failing while trying to install ansible. I just merged in #88 to let Ansible build with wheel like it's trying to do. It installs fine either way on Raspberry Pi OS 32-bit, but it might make a difference on aarch64.

Can you try running the quick install command again now that I've updated it?

```bash
curl -sSL https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | bash -
```

danmed, 2020-07-25:
I've just done

    sudo apt-get install python3-dev

And it seems to be potentially working.. bear with me.

mtlynch, 2020-07-25:
It looks like you're still pulling the old Ansible role. Can you give the new script a try?

```bash
curl -sSL https://raw.githubusercontent.com/mtlynch/tinypilot/install-wheel/quick-install | bash -
```

It's idempotent, so running it multiple times should be fine.

danmed, 2020-07-25:
> ```shell
> https://raw.githubusercontent.com/mtlynch/tinypilot/install-wheel/quick-install
> ```

Giving it a go now.. added my changes in so it runs for me.

danmed, 2020-07-25:
So, it was all going great.. no bdist errors.. then this happened : 

`> {"changed": false, "msg": "Destination /boot/config.txt does not exist !", "rc": 257}

PLAY RECAP *****************************************************************************************************************************************************************************************************
localhost                  : ok=24   changed=0    unreachable=0    failed=1    skipped=9    rescued=0    ignored=0

Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
[... 69 lines cut ...]
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0x7f96d0db70>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
`

mtlynch, 2020-07-25:
Hmm, I'm not sure what's going on there, but I notice that's Python3.5 whereas I've done all my testing on Python 3.7.3 (which is what ships with RasPi OS).

danmed, 2020-07-26:
> 
> 
> Hmm, I'm not sure what's going on there, but I notice that's Python3.5 whereas I've done all my testing on Python 3.7.3 (which is what ships with RasPi OS).

I installed 3.7.1 (or maybe .8 not sure lol) but still not working.. if i get time on Monday i'll try and start from scratch but this may just be a no go at the moment. thank you for all your help though.. the main reason i wanted to use one of thes (other than having one already) is that it has 2 options for power.. a normal barrel or a micro usb.. so i was going to use the barrel to free up the micro usb otg port.. (not that i've gotten far enough to check itis actually otg)

will report back.

Tampa, 2020-07-26:
The C4 definitely has an otg enabled port, even says so in the tech specs on the hardkernel website, so that is what I am getting along with an otg enabled 4port usb hub to attempt to get multiple machines hooked up to it. I hope multiple kvm connections are supported? Else I'll stuff it into containers of some sort. The nearest 4 port kvm switch is 630€, hardware for the diy version should be less than half that. Will monitor this and report what I find once I get the hardware delivered.

@danmed If you need help with tinkering, let me know, so far I have gotten almost anything to somehow run on my odroid

danmed, 2020-07-27:
> 
> 
> The C4 definitely has an otg enabled port, even says so in the tech specs on the hardkernel website, so that is what I am getting along with an otg enabled 4port usb hub to attempt to get multiple machines hooked up to it. I hope multiple kvm connections are supported? Else I'll stuff it into containers of some sort. The nearest 4 port kvm switch is 630€, hardware for the diy version should be less than half that. Will monitor this and report what I find once I get the hardware delivered.
> 
> @danmed If you need help with tinkering, let me know, so far I have gotten almost anything to somehow run on my odroids

Would love some help :)

mtlynch, 2020-07-27:
I ordered an ODROID C4 so I'll be able to help with this when it arrives next week.

danmed, 2020-07-28:
I've ordered a pi 4 and all the bits as i really could do with this working now, however if the odroid C2 can work, then it would be very handy to get going for my other system.

mtlynch, 2020-08-01:
I was able to repro this on an ODROID C4. I should have a fix this week.

lamarios, 2020-08-17:
As stated in #135  I'm using an [up squared board](https://up-board.org/upsquared/specifications/), is there anything I can do to help ?
I didn't use the Ansible script though (I have other stuff running on this board) but I managed to install everything without error by following de dev guide, and reading through the ansible repositories and setting up nginx my self and tinypilot and the video signal work fine.

On my guest (the server I want to control) I got this on dmesg:
```
[625161.432990] input: tinypilot Generic USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1D6B:0104.000C/input/input35
[625161.500154] hid-generic 0003:1D6B:0104.000C: input,hidraw0: USB HID v1.01 Keyboard [tinypilot Generic USB Keyboard] on usb-0000:00:14.0-7/input0

```

But somehow the keyboard input don't seem to work. What can I do to help work on this ?

mtlynch, 2020-08-17:
@lamarios - I believe the missing piece on other OSes is loading the dwc2 kernel module. On Pi OS, we can load it by just [adding lines](https://github.com/mtlynch/ansible-role-tinypilot/blob/176c2b76ce7aa274a8844d747a22969ec10a422d/tasks/install_usb_gadget.yml#L2-L12) to `/boot/config.txt` and `/etc/modules`, but I haven't dug into it to figure out the process on other OSes.

There might be additional work required to load the [USB gadget API](https://www.kernel.org/doc/html/v4.13/driver-api/usb/gadget.html), but I'm not sure if that comes with the Linux kernel by default or if there's something necessary we have to do to enable it on other OS builds.

lamarios, 2020-08-18:
I'm using Ubuntu 18.04 (x86_64) bits it seems that the dwc driver loaded is dwc3 though, I'm not sure it makes a difference. I tried to load dwc2 but it doesn't seem to do anything.

```
cat /sys/kernel/config/usb_gadget/g1/UDC 
dwc3.0.auto
```

lamarios, 2020-08-18:
Ok, i'm an idiot, it works perfectly fine, I was just running tiny pilot as my user and didn't have permission to /dev/hidg0 running tinypilot as root fixes the issue.

Thanks !

mtlynch, 2020-08-18:
Thanks for the donation, @lamarios!

Did you have to do anything special to get `/dev/hidg0` to appear under Ubuntu 18.04 or did it just work right out of the box with the same TinyPilot scripts?

lamarios, 2020-08-19:
It worked straight away with what's in the repo.

WarheadsSE, 2021-02-15:
I've had to do some tweaking due to poor detection of CPU type in the installation scripting (Ansible), but I've manged to get all variants of the [LibreComputer Tritium H3](https://libre.computer/products/boards/all-h3-cc/) working with [armbian](https://www.armbian.com/tritium-h3/).

mtlynch, 2021-02-15:
@WarheadsSE awesome! Could you share the configuration you used?

WarheadsSE, 2021-02-15:
@mtlynch That's the plan. I have to dig it all out of the nodes :sweat_smile:

WarheadsSE, 2021-02-15:
[This gist](https://gist.github.com/WarheadsSE/f90359e0f6955478c806ecbd04a643a3) covers the items needed to enabled, but I haven't yet found time to document a walk-through of how to enabled device tree overlay.

WarheadsSE, 2021-02-15:
Said gist updated (plus another) for LibreComputer Tritium

- [Parts list](https://gist.github.com/WarheadsSE/729633f0ecc6e5453f14485f2d09c87f)
- [LibreComputer Tritium OTG with armbian](https://gist.github.com/WarheadsSE/f90359e0f6955478c806ecbd04a643a3)

WarheadsSE, 2021-02-17:
Board | OS | Status
-- | -- | --
LibreComputer Tritium H5 | Armbian_21.02.1_Tritium-h5_buster_current_5.10.12 | [Enable the OTG functionality](https://gist.github.com/WarheadsSE/f90359e0f6955478c806ecbd04a643a3)

WarheadsSE, 2021-02-17:
Board | OS | Status
-- | -- | --
LibreComputer Tritium H5 | Armbian_21.02.1_Tritium-h5_buster_current_5.10.12 | [TinyPilot on LibreComputer Tritium](https://gist.github.com/WarheadsSE/f90359e0f6955478c806ecbd04a643a3)

Here's a new entry for above @mtlynch :+1:

mtlynch, 2021-02-17:
@WarheadsSE - added, thanks!

closed 2021-08-12 by mtlynch