tiny-pilot/tinypilot #80: Add support for power/reset from DIY-IPMI

Fmstrat, 2020-07-24:
Leverage the reboot/power button functionality from https://github.com/Fmstrat/diy-ipmi

Relay board: https://www.amazon.com/dp/B0057OC5WK
GPIO cables: https://www.amazon.com/gp/product/B01BV2A54G
Wiring diagram: https://raw.githubusercontent.com/Fmstrat/diy-ipmi/master/Resources/Diagram.png

Tsaukpaetra, 2021-01-14:
IMO relays for that are way overkill, they're sense lines (I think 5v falling-edge trigger?), so you should be able to use optocouplers to achieve this effect. 🤔

wesleyscaldwell, 2021-10-12:
----- copying my comments from 803 -----

I think ATX pin is the better option.
But that being said, maybe you can sell an add on relay module that connects to the pi.

Because for more permanent locations it's worth it to hook up the ATX pins, but for ad-hoc support locations a simple relay outlet might be easier.

db39, 2023-07-31:
A [user on the forum](https://forum.tinypilotkvm.com/-722/make-tiny-pilot-have-complete-power-on-and-reset-via-cables-to-the-motherboard) is also interested in this functionality.

DeviousFusion, 2023-08-13:
Would really appreciate if this was integrated in TinyPilot. While external ON/OFF switches are a good option, I think its best to use ATX pins. LED status will be helpful to know the system state.

binooetomo, 2023-12-05:
My proposal for this feature is :

1. TinyPilot send broadcast message via RS232.   
Message format is only 2 char <address><command>   
where :
A. 'address' is : A-Z
B. 'Command': A=Power Off, B= Power On, C=Reboot, 'Status'

2. At the managed node, user must have some kind of micro controller (arduino?) that interpret this message and do the job (if the address is it's address).

3. Network:
a. TX pin from TinyPilot connected to RX pin of micro controller on all managed node, and
b. RX pin from TinyPilot connected to TX pin of micro controller on all managed node.

Note:
1. Set the baud rate as low as posible ... message is very short no need for extensive speed, and we got stability 
2. With this scenario, user can also extend the range directly or just make a simple additional device to make it talk different protocol (ie: RS485)
All we ask is just TinyPilot 'say something' via it's RS232.