tiny-pilot/tinypilot #83: Issue: Keyboard connection error

KuroSaru, 2020-07-24:
Hardware: RaspberrryPI 4B 8GB
OS: RaspberryOS 64bit

Display work flawlessly. installation script never installs the files to setup the hid, if installed manually get keyboard connection error message. Generates a Error -71 on a Linux host the tinypilot is connected to.

mtlynch, 2020-07-24:
Thanks for reporting this, @KuroSaru and sorry for the issues. Would you be able to share a [gist](https://gist.github.com/) with the output when you run the install command?

```bash
curl -sSL https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | bash -
```

KuroSaru, 2020-07-24:
So i fixed my issues manually.  But i will run script to get output still as now i know my issue.

1. CORS was blocking sending inputs
2. Had to manually insatll the usb-gadget service no idea why it wasn't created.

-------------------------------------------------------------------
Output:
`
pi@frank:/opt $ curl -sSL https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | bash -
+ set -e
+ set -u
+ '[' -f /etc/nginx/sites-enabled/KVMPi.conf ']'
++ mktemp -d
+ pushd /tmp/tmp.IYXuNiY6bZ
/tmp/tmp.IYXuNiY6bZ /opt
+ sudo apt-get install -y python3-venv
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-venv is already the newest version (3.7.3-1).
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ python3 -m venv venv
+ . venv/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/tmp/tmp.IYXuNiY6bZ/venv
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ PATH=/tmp/tmp.IYXuNiY6bZ/venv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ export PATH
++ '[' -n '' ']'
++ '[' -z '' ']'
++ _OLD_VIRTUAL_PS1=
++ '[' 'x(venv) ' '!=' x ']'
++ PS1='(venv) '
++ export PS1
++ '[' -n /bin/bash -o -n '' ']'
++ hash -r
+ pip install ansible==2.9.10
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible==2.9.10
  Using cached https://files.pythonhosted.org/packages/03/4f/cccab1ec2e0ecb05120184088e00404b38854809cf35aa76889406fbcbad/ansible-2.9.10.tar.gz
Collecting PyYAML (from ansible==2.9.10)
  Using cached https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f9501676c988a5411ed5572f35d1beff9cae07d321512c/PyYAML-5.3.1.tar.gz
Collecting cryptography (from ansible==2.9.10)
Collecting jinja2 (from ansible==2.9.10)
  Using cached https://files.pythonhosted.org/packages/30/9e/f663a2aa66a09d838042ae1a2c5659828bb9b41ea3a6efa20a20fd92b121/Jinja2-2.11.2-py2.py3-none-any.whl
Collecting six>=1.4.1 (from cryptography->ansible==2.9.10)
  Using cached https://files.pythonhosted.org/packages/ee/ff/48bde5c0f013094d729fe4b0316ba2a24774b3ff1c52d924a8a4cb04078a/six-1.15.0-py2.py3-none-any.whl
Collecting cffi!=1.11.3,>=1.8 (from cryptography->ansible==2.9.10)
  Using cached https://files.pythonhosted.org/packages/05/54/3324b0c46340c31b909fcec598696aaec7ddc8c18a63f2db352562d3354c/cffi-1.14.0.tar.gz
Collecting MarkupSafe>=0.23 (from jinja2->ansible==2.9.10)
  Using cached https://files.pythonhosted.org/packages/b9/2e/64db92e53b86efccfaea71321f597fa2e1b2bd3853d8ce658568f7a13094/MarkupSafe-1.1.1.tar.gz
Collecting pycparser (from cffi!=1.11.3,>=1.8->cryptography->ansible==2.9.10)
  Using cached https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl
Building wheels for collected packages: ansible, PyYAML, cffi, MarkupSafe
  Running setup.py bdist_wheel for ansible ... error
  Complete output from command /tmp/tmp.IYXuNiY6bZ/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-y_li2r_p/ansible/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-xdizewp_ --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
[... 210 lines cut ...]

TASK [mtlynch.tinypilot : get TinyPilot repo] **********************************
changed: [localhost]

TASK [mtlynch.tinypilot : create TinyPilot virtualenv] *************************
changed: [localhost]

TASK [mtlynch.tinypilot : fix TinyPilot folder permissions] ********************
ok: [localhost]

TASK [mtlynch.tinypilot : install TinyPilot as a service] **********************
changed: [localhost]

TASK [mtlynch.tinypilot : enable systemd TinyPilot service file] ***************
ok: [localhost]

RUNNING HANDLER [mtlynch.tinypilot : reload TinyPilot systemd config] **********
ok: [localhost]

RUNNING HANDLER [mtlynch.tinypilot : restart TinyPilot service] ****************
changed: [localhost]

PLAY RECAP *********************************************************************
localhost                  : ok=36   changed=5    unreachable=0    failed=0    skipped=14   rescued=0    ignored=0
`

mtlynch, 2020-07-24:
>CORS was blocking sending inputs

Ah, right! It assumes the hostname is `tinypilot`. I'll fix that.

>Had to manually insatll the usb-gadget service no idea why it wasn't created.

Oh, that's strange. Did the install script report any errors the first time you ran it?

KuroSaru, 2020-07-24:
> > Had to manually insatll the usb-gadget service no idea why it wasn't created.
> 
> Oh, that's strange. Did the install script report any errors the first time you ran it?
Sadly it did not. only thing it said as shown above skipping. I checked for /opt/enable-rpi-hid and /lib/systemd/system/usb-gadget.service neither were created.  Might be because im using the raspberrypi OS 64bit beta? and scripts didn't detect it correctly.

mtlynch, 2020-07-24:
>Might be because im using the raspberrypi OS 64bit beta? and scripts didn't detect it correctly.

Oh, that's likely it. I haven't tested on 64-bit yet, so I think you're the first to do this!

But given that you got it by installing manually, I should be able to get it working with some small adjustments to the Ansible script to correct for whatever it's missing.

My guess is that the Ansible architecture shows up as something that doesn't start with `armv` on Pi 64. Is it aarch64?

https://github.com/mtlynch/ansible-role-tinypilot/blob/5329a9b5fd18c7254f2b654c462ac9026c831204/tasks/main.yml#L4

KuroSaru, 2020-07-24:
Yeah it is aarch64. 
Sidenote: also had to manually run
```
sudo apt-get install libffi-dev
```

mtlynch, 2020-07-24:
Ah, thanks for the note. I created a separate bug to track 64-bit compatibility:

https://github.com/mtlynch/ansible-role-tinypilot/issues/19

closed 2020-07-24 by mtlynch

bgulla, 2020-07-29:
> Yeah it is aarch64.
> Sidenote: also had to manually run
> 
> ```
> sudo apt-get install libffi-dev
> ```

I am having the same problem that you are. In addition to the libffi-dev install, what exactly did you have to enable to get the usb-gadget working? I am seeing `inypilot enable-rpi-hid[795]: /opt/enable-rpi-hid: line 30: echo: write error: Device or resource busy`.

KuroSaru, 2020-07-29:
> I am having the same problem that you are. In addition to the libffi-dev install, what exactly did you have to enable to get the usb-gadget working? I am seeing `inypilot enable-rpi-hid[795]: [...]

You have to install the usb-gadget manually. Generally was adding the service manually. I think the author has a patch for that now or is working on it. As the deploy script for the usb-gadget never triggered on arm64 builds.

PR #84 (2020-07-24): Eliminate assumption that hostname is tinypilot
Change CORS_ORIGIN to default to the local machine's hostname rather than assume that the machine is called 'tinypilot'.

Fixes #83