tiny-pilot/tinypilot #85: error: invalid command 'bdist_wheel'

Savasci, 2020-07-24:
Hardware: Raspberry Pi 4 Model B Rev 1.1, Revision a03111
OS: Raspberry Pi OS (32-bit) Lite, up to date

Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux

Somehow the quick-install fails with 
`fatal: [localhost]: FAILED! => {"changed": false, "cmd": "/opt/tinypilot/venv/bin/pip list --format=freeze", "msg": "[Errno 2] No such file or directory", "rc": 2}`

I suspect ansible failing already:
```
Building wheels for collected packages: ansible
  Running setup.py bdist_wheel for ansible ... error
  Complete output from command /tmp/tmp.3iDxsDrTiL/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-syk1vi37/ansible/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-a3rj78f2 --python-tag cp37:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for ansible
  Running setup.py clean for ansible
Failed to build ansible
```

Full gist at 
https://gist.github.com/Savasci/7f7de61a1ff684c16d19dadaaa980abb

mtlynch, 2020-07-24:
Thanks for reporting this!

The `bdist_wheel` failure is normal on the Pi.

That's strange that it's dying on that step. Can you try re-running the command? Everything in the install script is idempotent, so it's fine to re-run it.

Savasci, 2020-07-24:
Thank you for having a look! I ran this a few times to check if this is one of the limiting steps and it seems it is. Line 5 of the gist shows, that this was one of my later re-reruns, the error stays the same.

Savasci, 2020-07-24:
I've started from scratch now to see if one of the package updates is causing this issue, I'm restarting from a clean 2020-05-27-raspios-buster-lite-armhf.img to check.

Savasci, 2020-07-24:
Ok, so, on
`Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux`
the quick-setup works without issues, the bdist_wheel failure still being present, as you said. My final stage on a clean a clean 2020-05-27-raspios-buster-lite-armhf:

```
PLAY RECAP *********************************************************************
localhost                  : ok=46   changed=31   unreachable=0    failed=0    skipped=8    rescued=0    ignored=0
```

apt-get upgrade is about to upgrade the following packages:
`ca-certificates firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek libgnutls30 libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel raspi-config rpi-eeprom rpi-eeprom-images`
from which at least one of them seems to brick the quick-install.

mtlynch, 2020-07-24:
Hmm, that's strange. I'll try to repro on my side, but let me know if you discover anything more about why it's falling over on the virtualenv step.

Savasci, 2020-07-24:
Thanks! Sure, I'll add if I find out, why that one happens.

Ah, upgrading after installing via quick-install works fine, currently running on
`Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux`

mtlynch, 2020-08-10:
Fixed in https://github.com/mtlynch/ansible-role-tinypilot/pull/32

closed 2020-08-10 by mtlynch