tiny-pilot/tinypilot #92: Wi-Fi hotspot support

bderenzo, 2020-07-27:
It could be nice to add an hotspot option.
TinyPilot could then become standalone and portable.
* Connect the USB and HDMI to server
* Connect the computer via WIFI on TinyPilot's hotspot 
* Access server remotely

mtlynch, 2020-07-27:
Interesting idea! I hadn't thought of that.

Gh0stWalk3r, 2020-07-27:
This could be combined with a physical toggle button to create a WPS like connectivity/visibility.
e.g. on push of the button the WIFI SSID will be visible for 5min or until successful connectivity.

somik123, 2020-08-01:
> * Connect the USB and HDMI to server
> * Connect the computer via WIFI on TinyPilot's hotspot
> * Access server remotely
* Connect the Pi 4 to your powerbank to power it
and we have a fully portable KVM type thing. Hope to see this in future maybe.

rvdwegen, 2020-08-08:
> > * Connect the USB and HDMI to server
> > * Connect the computer via WIFI on TinyPilot's hotspot
> > * Access server remotely
> 
> * Connect the Pi 4 to your powerbank to power it
>   and we have a fully portable KVM type thing. Hope to see this in future maybe.

- Add a e-paper hat to display the ssid, password and IP.

rvdwegen, 2020-08-19:
> > > * Connect the USB and HDMI to server
> > > * Connect the computer via WIFI on TinyPilot's hotspot
> > > * Access server remotely
> > 
> > 
> > 
> > * Connect the Pi 4 to your powerbank to power it
> >   and we have a fully portable KVM type thing. Hope to see this in future maybe.
> 
> * Add an e-paper hat to display the ssid, password and IP.

So setting a Pi up as AP wasn't much of a hassle. Though proper documentation for 5Ghz support is spotty. I can say that it essentially works. I do appear to have some stability issues but I'm currently running without heatsinks so I'll have to wait for other parts to come in from China.

bderenzo, 2020-08-28:
@rvdwegen Indeed installating docker just to configure wifi access point is a bit overkill.

I received all the parts today and configure everything. It works like a charm (without increasing latency).
Here is what I've done with root 

```
# Disable unused services
systemctl stop alsa-restore.service
systemctl stop alsa-state.service
systemctl stop triggerhappy.socket
systemctl stop bluetooth.service
systemctl stop triggerhappy.service
systemctl stop avahi-daemon.service
systemctl stop avahi-daemon.socket

systemctl mask alsa-restore.service
systemctl mask alsa-state.service
systemctl mask triggerhappy.socket
systemctl mask bluetooth.service
systemctl mask triggerhappy.service
systemctl mask avahi-daemon.service
systemctl mask avahi-daemon.socket

# Install wifi access point and network management softward
apt-get -y install dnsmasq hostapd

# Configure a static ip on wlan interface
cat << EOF >> /etc/dhcpcd.conf

# Wlan config
interface wlan0
static ip_address=192.168.5.1/24
nohook wpa_supplicant
EOF

# Configure wifi acces point (SSID RaspiKVM, password RaspiKVM)
cat << EOF > /etc/hostapd/hostapd.conf
interface=wlan0
ssid=NameOfNetwork
hw_mode=g
channel=6
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
rsn_pairwise=CCMP
ssid=RaspiKVM
wpa=2
wpa_passphrase=RaspiKVM
wpa_passphrase=AardvarkBadgerHedgehog
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
EOF

# Set wifi access point default config file
sed 's|.*DAEMON_CONF=.*$|DAEMON_CONF="/etc/hostapd/hostapd.conf"|g' -i /etc/default/hostapd

# Configure DHCP
cat << EOF >> /etc/dnsmasq.conf
interface=wlan0
dhcp-range=192.168.5.2,192.168.5.200,255.255.255.0,24h
EOF

# Enable everything
systemctl unmask hostapd
systemctl enable hostapd
systemctl enable dnsmasq

# Reboot
reboot

# Enjoy ;)
```

It works with only HDMI and one USB 3.0 A to C cable.
But sometimes there are "Under-voltage" error. An Y-cable can solve the problem.

rvdwegen, 2020-08-29:
@bderenzo How is the performance in g mode?

mtlynch, 2024-01-04:
We've now published official instructions for [using TinyPilot as a Wi-Fi hotspot](https://tinypilotkvm.com/faq/wifi-ap).

closed 2024-01-04 by mtlynch