tiny-pilot/tinypilot #93: Pi Zero W

peterberbec, 2020-07-28:
The lower power requirements, smaller size, and lower heat of the Pi0 allow for an interesting possibility. The 0 could be totally powered through the OTG cable and with a otg adapter, the hdmi capture card could still be attached.

I'm picking up a P and a OTG adapter. I will report back.

peterberbec, 2020-07-30:
It's here and I'm doing the install. Updates later. Pi 0 is slow, but that shouldn't matter since the overhead is minimal, right?

[image]

peterberbec, 2020-07-30:
Works

[image]

mtlynch, 2020-07-30:
Nice! Thanks for testing that, @peterberbec!

closed 2020-07-30 by mtlynch

rvdwegen, 2020-08-03:
@peterberbec what was the video performance like?

peterberbec, 2020-08-03:
It was definately NOT under 200ms of lag, but usable.

I'm testing out a FriendlyElec NanoPi Meo mext

mtlynch, 2020-08-03:
@peterberbec would you be able to share the name/link of the USB cable you're using with the Pi?

peterberbec, 2020-08-03:
The microUSB to regular USB and charging port cable?

That's a microUSB OTG/charge cable.

https://www.amazon.com/dp/B07FY9Z9GD

rvdwegen, 2020-08-06:
Hey, did you have to make any changes or was it plug and play on the Pi zero?

peterberbec, 2020-08-06:
Plug, curl | bash, play

Peter Berbec
(917) 213-2594

rvdwegen, 2020-08-08:
And the keyboard works too?

somik123, 2020-08-08:
> And the keyboard works too?

Doubt it. Pi zero has only 1 USB that supports data (other one is power only, no data). So it can either behave as host (for the HDMI adapter) or as OTG (for keyboard input). Since he is using the only data port for HDMI to usb adapter, no keyboard input is available.

[image]

peterberbec, 2020-08-09:
I could have connected a hub to the otg, or a Bluetooth keyboard, but I
didn't. I can test such a setup if their is interest. I was doing it as a
test of the Zero platform's viability, and the speed makes it not useful to
me.

somik123, 2020-08-09:
@peterberbec 
> I could have connected a hub to the otg, or a Bluetooth keyboard, but I didn't. I can test such a setup if their is interest. I was doing it as a test of the Zero platform's viability, and the speed makes it not useful to me.

I did connect it to a usb hub. Only the video capture works. You need to connect it directly to PC without any hub to get the keyboard input to work, but then video wont work. So you have to choose between the two. You dont have 2 usb controllers to make one act as host and the other device, so both video and keyboard wont work at the same time on zero for now.

peterberbec, 2020-08-09:
Ah. My mistake. I thought you meant a keyboard to the Pi, not the otg
keyboard

somik123, 2020-08-09:
> Ah. My mistake. I thought you meant a keyboard to the Pi, not the otg keyboard

Ya, i meant the device mode where the Pi sends keyboard input to the PC.

mtlynch, 2020-08-10:
One idea for the Pi Zero would be to use the CSI bridge to capture HDMI, like this one:

https://www.ebay.com/itm/For-Raspberry-Pi-HDMI-To-CSI-2-C779-Bridge-Adapter-Interface-Module-1080p-25fps/203009279885

That capture card works with TinyPilot, though the Ansible role requires a few tweaks to handle the HDMI to CSI-2 bridge instead of the HDMI dongle.

peterberbec, 2020-08-10:
Purchased. I will report back

peterberbec, 2020-09-03:
> One idea for the Pi Zero would be to use the CSI bridge to capture HDMI, like this one: [...]

I purchased that exact adapter, and it works. Keyboard input does not seem to be functioning. What steps should I go through to troubleshoot g_hid?

peterberbec, 2020-09-03:
> I purchased that exact adapter, and it works. Keyboard input does not seem to be functioning. What steps should I go through to troubleshoot g_hid?

Step 1: Use a usb cable that supports data, not just charging.

It's working!

rvdwegen, 2020-09-03:
Excellent news!

This could make for a fairly compact package if one were to take a Lusya bridge from China (same screw holes as pi zero) and one of those cheapo USB stick conversion PCB's.

https://ae01.alicdn.com/kf/Hc7195b14d18d425f8a7b76b48621d9a66.jpg
https://imgaz.staticbg.com/thumb/large/oaupload/banggood/images/87/A8/f81af760-9f26-4101-b1f9-0e25273cd1fa.jpg

somik123, 2020-09-03:
[image]

[image]

somik123, 2020-09-03:
From top:
HDMI to CSI bridge, 
Raspberry Pi Zero W, 
1000mAh battery module.

espoelstra, 2020-11-08:
@somik123 that is awesome!!! I had a similar idea but hadn't thought to put a battery in there to allow the Pi0 to still function (probably for HOURS) through host reboots or brief power outages.

I think I have a CSI bridge from that same company but with a larger footprint closer to the Pi0 size. Did you have to do anything special to get it recognized by TinyPilot?

somik123, 2020-11-09:
> @somik123 that is awesome!!! I had a similar idea but hadn't thought to put a battery in there to allow the Pi0 to still function (probably for HOURS) through host reboots or brief power outages.
>
> I [...]

Yes, there was a bit of configuration change involved. I can't find that thread now though. You need to change the startup script for the video capture software to ensure it gets the data from this csi bridge instead of looking for a USB device.

mtlynch, 2020-11-09:
I'm working on handling the HDMI to CSI chip configuration natively in the TinyPilot installer. For now, you can do this:

```bash
export TINYPILOT_INSTALL_VARS="ustreamer_encoder=omx ustreamer_capture_device=tc358743 ustreamer_workers=3 ustreamer_drop_same_frames=30 ustreamer_use_dv_timings=yes ustreamer_format=uyvy"
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

I'm working on that this week, though, so expect a better solution within days.

swbullis, 2020-11-21:
### The right way
I, unfortunately, didn't see this issue yesterday when I was testing things out, however, I want to call out @somik123 solution above as the way forward. 

Having a dedicated power supply (battery) solves the problem of the pi resetting during reboots.  And installing a ZeroStem ( I bought mine here https://www.adafruit.com/product/3945) that you can solder onto the board makes for a very compact solution.

And of course, to get the video into the raspberry pi you will need something like this https://www.ebay.com/itm/For-Raspberry-Pi-HDMI-To-CSI-2-C779-Bridge-Adapter-Interface-Module-1080p-25fps/203009279885 

this is because the Pi Zero only has one USB port so it can send data on that port  (gadget mode) or receive data on that port (regular mode) all of this is said higher up in other comment but I am including it here for thoroughness.

### The Easy way
Ultimately what I ended up doing was ordering a Pi 4

### The I can't wait I need it now way

However as stop gap until the correct parts arrived. I took the following actions which might be useful if you happen to have 2 Pi Zero W's available to you.

Install TinyPilot on Both Pi's and setup 1 to Receive Video using the HDMI to USB Adapter (This is just plugging in the adapter)
You should be able to test that it's working by navigating to the IP address for that device.

Setup the other as normal this one will be used to emulate the keyboard and mouse.  (No video adapter connected)

Plug both into your target computer and navigate to the IP address of Pi 2 (Keyboard and mouse one).  Go into your developer tools and change the image address to point to the IP address of the other Tiny Pilot.

It should look something like this

```
<div class="screen-wrapper" cursor="crosshair">
    <input id="mobile-keyboard-input" autocapitalize="off" type="text">
    <img id="remote-screen-img" src="http://192.168.1.111/stream?advance_headers=1">
  </div>
```

Hope this helps someone else!

peterberbec, 2020-11-21:
Thinking about this now, I wonder why a Bluetooth keyboard & mouse didn't
occur to me. Csi for hdmi input, usb for HID to "main" computer, the pi's
Micro (?) hdmi for video, and Bluetooth keyboard/mouse.

mtlynch, 2020-11-21:
I improved handling HDMI to CSI capture devices. The best way to configure for it is now this:

```bash
export TINYPILOT_INSTALL_VARS="ustreamer_capture_device=tc358743"
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

I did some experimenting today with the Pi Zero:

https://twitter.com/tinypilotkvm/status/1329881175847002113

It seems way too slow to be a feasible platform for TinyPilot. Are others just dialing down the resolution to 640x480 or something tiny?

tristan-k, 2021-07-28:
Did anyone try to use the bluetooth of rpi zero w as an iput method for tinypilot? I'm still trying to figure out a way for input on the zero w without buying a csi bridge.