tiny-pilot/tinypilot #96: Windows BitLocker Support

thetanil, 2020-07-29:
It seems that bitlocker (ms drive encryption) boots into some video mode which is not captured by the recommended (non-branded) HDMI capture device (or lost in the encoding).

Additionally the bitlocker screen times out and powers down about the time that the pi actually boots through to UI available. Just wanted to document somewhere that the bitlocker timeout can be disabled with the following command:

    bcdedit /set {bootmgr} bootshutdowndisabled 1

Windows Details:

    Windows Edition: Win 10 Enterprise 
    Version: 1809
    OS Build: 17763.1282

somik123, 2020-07-31:
@thetanil FYI, you don't need to power on the Pi along with the computer. You should power on the Pi (or keep it on), get to the browser UI, then power on your PC.

mtlynch, 2021-03-09:
I believe BitLocker deliberately doesn't display to a secondary monitor on certain systems, so it's not possible for TinyPilot to overcome this.

closed 2021-03-09 by mtlynch