tiny-pilot/tinypilot #99: Restrict CORS origins for socket.io connections

mtlynch, 2020-07-29:
In #98, I removed strict checking for CORS requests. My reasoning is that it's not providing meaningful protection at the moment, but it's creating significant usability issues (see #97).

CORS origin checking currently does not much protection because:

1. The user base of TinyPilot is so small that CSRF attacks are highly unlikely.
1. Even with CORS origin checking, third-party origins can still send requests to the `/socket.io` route from the user's browser. CORS enforcement prevents them from reading responses, but there's nothing useful in the responses anyway.

We should revisit this decision in the future to ensure that we're not sacrificing real protection by dropping origin checking, even if (1) is no longer true.

Bramzor, 2020-08-08:
True, I think CORS has almost no benefit especially since there is no authentication and traffic is not going over HTTPS. So I'm more worried about passwords that are  possibly going over an unencrypted connection.

If HTTPS would be added (not easy to do and takes additional resources), activating CORS would be a no brainer as using the correct hostname becomes a requirement anyway.

mtlynch, 2021-03-12:
Reopening as the previous fix caused a regression. When users accessed TinyPilot from a different URL, it broke the websocket connection.

Need to investigate more how to make SocketIO cooperate.

mtlynch, 2021-03-12:
From tinkering some more, it seems to be related to nginx. Connecting directly to TinyPilot without nginx works as expected.

I need to review the nginx config here:

https://flask-socketio.readthedocs.io/en/latest/#using-nginx-as-a-websocket-reverse-proxy

PR #559 (2021-03-09): Remove CORS workaround for Flask-SocketIO
I originally added wildcard support so that users could access TinyPilot from addressess like http://tinypilot or http://10.0.0.123 and it was breaking when CORS was hardcoded to a specific address. But the better fix is to just remove the CORS handling altogether and accept Flask-SocketIO's default behavior of same-origin, which will work regardless of whether the user accesses by IP, hostname, or even a remapped hostname that the server doesn't know about.

Fixes #99
PR #587 (2021-03-18): Remove wildcard from CORS allowed origins.
Based on the [fix applied to the nginx config](https://github.com/mtlynch/ansible-role-tinypilot/pull/113), we can now remove the wildcard from the CORS allowed origins.

Fixes #99
closed 2021-03-18 by jdeanwallace