tiny-pilot/tinypilot #103: History box expands even if unticked

danmed, 2020-07-31:
Tested in Firefox and Chrome (both latest) on Windows 10.

Navigate to the page, untick the history, start typing.. the box expands anyway.

mtlynch, 2020-07-31:
Thanks! I'll see if I can reproduce this.

somik123, 2020-08-01:
Confirmed issue on Vivaldi as well.

Change

```
function onDisplayHistoryChanged(evt) {
  if (evt.target.checked) {
    document.getElementById("recent-keys").style.visibility = "visible";
  } else {
    document.getElementById("recent-keys").style.visibility = "hidden";
    limitRecentKeys(0);
  }
}
```

on **app.js** to

```
function onDisplayHistoryChanged(evt) {
  if (evt.target.checked) {
    document.getElementById("recent-keys").style.display = "";
  } else {
    document.getElementById("recent-keys").style.display = "none";
    limitRecentKeys(0);
  }
}
```

Or, better yet, use the function that you already declared:

```
function onDisplayHistoryChanged(evt) {
  if (evt.target.checked) {
    showElementById("recent-keys");
  } else {
    hideElementById("recent-keys");
    limitRecentKeys(0);
  }
}
```

PR #110 (2020-08-03): Prevent history area from expanding when history is disabled
This fixes an issue where the key history area would expand as the user types keys even if they had disabled key history. The keys wouldn't display, but the empty space expanding was distracting/confusing.

Fixes #103
commit 7e67d909 (2020-08-03): Prevent history area from expanding when history is disabled (#110)

This fixes an issue where the key history area would expand as the user types keys even if they had disabled key history. The keys wouldn't display, but the empty space expanding was distracting/confusing.

Fixes #103
closed 2020-08-03 by mtlynch