tiny-pilot/tinypilot #107: Suggestion: VNC server

manutenfruits, 2020-08-01:
It would be great if TinyPilot could expose a VNC server. I know it already exposes a web interface, but I use Apache Guacamole to manage my servers and it would be nice to expose TinyPilot using a standard protocol.

somik123, 2020-08-02:
I think the current focus for tinypilot is to emulate something like KVM-over-IP which you access through a browser directly.

If you install tinypilot to RaspbianOS desktop version and access the desktop through the VNC, tinypilot through the browser built in to the desktop, then you can access it. What do you think about that?

Bramzor, 2020-08-07:
I think the tricky part is that you need to get the keyboard input received by the VNC server and pass that to the virtual keyboard.
I actually had a similar requirement where I was looking for a way to have local keyboard (so connected to the raspberry pi) input passed to the virtual keyboard.

So if there is a way to pipe this information and combine it as it was a single input source (in another issue someone mentioned https://github.com/Blub/netevent ), I would think both of our use cases could be covered.