tiny-pilot/tinypilot #111: Add ability to paste

pwcazenave, 2020-08-04:
It'd be handy to be able to paste in the contents of the clipboard into the remote machine (this is particularly useful for long, strong passwords which are a nuisance to type in).

somik123, 2020-08-07:
+1 to this. Would be useful when we need to copy-paste linux commands as well for installation or configurations.
Maybe add a button that'll send whatever text is in the clipboard to the remote machine when clicked?

pwcazenave, 2020-08-07:
As it happens, I had a stab at implementing this just after I wrote the issue.

I'm not very good at javascript, but I managed to hack something together which seems to work for me. One of the main issues I feel like I'm not handling all that well is I have a hard-coded list of modifier keys. This obviously doesn't support keyboard layouts beyond what I'm using.

In any case, I've opened a pull request #113 with what I put together.

rvdwegen, 2020-09-03:
A thought, but Malduino/Rubber Ducky style keystroke injection functionality might be interesting too. The ability to store pre-prepared action scripts that emulate keystrokes and text input.

mtlynch, 2020-09-16:
Fixed in #113, though Firefox is still an open issue: #218

closed 2020-09-16 by mtlynch