tiny-pilot/tinypilot #112: Video is a broken image

rca, 2020-08-07:
I got my hobbyist bundle in the mail.  I fired it up and accessing the site results in a broken image.  Looking at the inspector, the error is:

```
GET http://localhost:8000/stream?advance_headers=1 404 (NOT FOUND)
```

However, I'm able to change the port in the URL above to `8001` in another browser window and the video works fine.

I ran `/opt/tinypilot/quick-install` and after `git log` in `/opt/tinypilot` on the Pi returns the following as the top commit: 

```
commit 7e67d9092d7f6e9e49b6520c85b448470ba8b82b (HEAD -> master, origin/master, origin/HEAD)
Author: Michael Lynch <mtlynch@users.noreply.github.com>
Date:   Mon Aug 3 17:02:48 2020 -0400

    Prevent history area from expanding when history is disabled (#110)

    This fixes an issue where the key history area would expand as the user types keys even if they had disabled key history. The keys wouldn't display, but the empty space expanding was distracting/confusing.

    Fixes #103
```

Also tried rebooting after running `quick-install`; same issue.

Thanks!

[image]

rca, 2020-08-07:
Nevermind, I should be accessing port 80, not 8000.

closed 2020-08-07 by rca

mtlynch, 2020-08-07:
Thanks for reporting this, @rca! Glad you got it working, but it's still useful to have a record of this in case anyone else runs into the same issue.

Were you were SSH'ing in and port-forwarding ports 8000 and 8001 back to localhost? I assume you've discovered this, but you can access port 80 without any SSH tunneling.

rca, 2020-08-07:
Yes, SSH and port forwarding, good guess! 🔮

For other reasons, I still need to port forward, but now I'm forwarding port 80 and it works as expected👍🏽

Thanks.