tiny-pilot/tinypilot #114: HDMI Bridge base on chip TC358743XBG

Bramzor, 2020-08-07:
I assume your project might also work together with an HDMI Bridge to CSI-2 (Raspberry Pi camera port) based on the popular TC358743XBG chip as HDMI IN? https://auvidea.eu/b101-hdmi-to-csi-2-bridge-15-pin-fpc/
Or the B102 in my case (Raspberry Pi zero 22 pin port)

Ordered one of these and will be able to test it in a few days. Only problem might be the increased resolution which might slow it down even more when combined with a raspberry pi zero. But maybe there is a way to lower the resolution or lower FPS.

mtlynch, 2020-08-07:
It should work, though you'll need to add/replace some of the [ustreamer settings](https://github.com/mtlynch/ansible-role-tinypilot/blob/260637d5a0914db36f25f47a2782b13bb921f513/meta/main.yml#L23) to:

```yaml
ustreamer_format: uyvy
ustreamer_encoder: omx
ustreamer_use_dv_timings: true
```

somik123, 2020-08-13:
@Bramzor do let us know as I'm also interested in this.

Bramzor, 2020-08-24:
Does not seem to work. /opt/ustreamer/ustreamer --persistent --dv-timings --format=uyvy --encoder=omx --workers=1 --quality=80 --desired-fps=30 --drop-same-frames=30 --last-as-blank=0
Unknown encoder type: omx; available: CPU, HW

Running it without the encoder line gives me:

/opt/ustreamer/ustreamer --persistent --dv-timings --format=uyvy --workers=1 --quality=80 --desired-fps=30 --drop-same-frames=30 --last-as-blank=0
-- INFO  [772.123      main] -- Installing SIGINT handler ...
-- INFO  [772.124      main] -- Installing SIGTERM handler ...
-- INFO  [772.124      main] -- Ignoring SIGPIPE ...
-- INFO  [772.124      main] -- Using internal blank placeholder
-- INFO  [772.125      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [772.125    stream] -- Using V4L2 device: /dev/video0
-- INFO  [772.125    stream] -- Using desired FPS: 30
================================================================================
-- INFO  [772.125    stream] -- Device fd=8 opened
-- INFO  [772.125      http] -- Starting HTTP eventloop ...
-- INFO  [772.125    stream] -- Using input channel: 0
-- INFO  [772.126    stream] -- Using TV standard: DEFAULT
-- ERROR [772.126    stream] -- Can't subscribe to V4L2_EVENT_SOURCE_CHANGE: Invalid argument
-- INFO  [772.126    stream] -- Device fd=8 closed

mtlynch, 2020-08-24:
Oh, looks like I need to fix something to make it recompile with the `WITH_OMX=1` option. See:

https://github.com/pikvm/ustreamer#building

Once it's properly built, this should work:

```bash
/opt/ustreamer/ustreamer \
    --format=uyvy \ # Device input format
    --encoder=omx \ # Hardware encoding with OpenMAX
    --workers=3 \ # Maximum workers for OpenMAX
    --persistent \ # Don't re-initialize device on timeout (for example when HDMI cable was disconnected)
    --dv-timings \ # Use DV-timings
    --drop-same-frames=30 # Save the traffic

```

stratus-ss, 2020-08-26:
I tried to make ustreamer with the following `make WITH_OMX=1 -j3` but when I try to run ustreamer with either the arguments from @mtlynch or @Bramzor I get the following:
```
ustreamer@raspberrypi:/home/pi$ /opt/ustreamer/ustreamer --format=uyvy --encoder=omx --workers=3 --persistent --dv-timings --drop-same-frames=30
-- INFO  [116.997      main] -- Installing SIGINT handler ...
-- INFO  [116.997      main] -- Installing SIGTERM handler ...
-- INFO  [116.997      main] -- Ignoring SIGPIPE ...
-- INFO  [116.997      main] -- Using internal blank placeholder
-- INFO  [116.997      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [116.998    stream] -- Using V4L2 device: /dev/video0
-- INFO  [116.998    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [116.998    stream] -- Device fd=8 opened
-- INFO  [116.998      http] -- Starting HTTP eventloop ...
-- INFO  [116.998    stream] -- Using input channel: 0
-- INFO  [116.998    stream] -- Using TV standard: DEFAULT
-- ERROR [116.998    stream] -- Can't subscribe to V4L2_EVENT_SOURCE_CHANGE: Invalid argument
-- INFO  [116.998    stream] -- Device fd=8 closed
-- INFO  [116.998    stream] -- Sleeping 1 seconds before new stream init ...

```

mtlynch, 2020-08-26:
@stratus-ss - you seem to be hitting a ustreamer bug:

https://github.com/pikvm/ustreamer/issues/38

stratus-ss, 2020-08-26:
thanks for the quick reply!

somik123, 2020-09-04:
any fix for it?

mtlynch, 2020-09-04:
I added instructions for configuring a HDMI to CSI capture chip here:

https://github.com/mtlynch/tinypilot/wiki/Installation-Options#example-csi-to-hdmi-capture-chip

It looks like one particular chip might be having issues, but that would be a bug in [the driver](https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=120702&start=675) not TinyPilot.

closed 2020-09-04 by mtlynch