tiny-pilot/tinypilot #129: Add option to reboot via a tasmota based plug

danmed, 2020-08-12:
I have coupled your project with a tasmota based smart plug.

this means i can remotely reboot the server.. but it would be nice to have that in the gui..

I can add it in myself... but it would probably be wiped from the index.html when i update..

would it be possible to have a configurable power on / off / restart button that i could put my smart plug IP address in to ?

the url format would be : 

* http://<ip>/cm?cmnd=Power%20On
* http://<ip>/cm?cmnd=Power%20off

and i guess a restart would be a delay between the two..

peterberbec, 2020-08-12:
There's also an open source project to control Tp-link smartplugs.
https://github.com/softScheck/tplink-smartplug

danmed, 2020-08-12:
> 
> There's also an open source project to control Tp-link smartplugs. https://github.com/softScheck/tplink-smartplug
> […](#)

it would be great to add options for different socket types... it lacks the finesse of a graceful shutdown.. but in the circumstances i can think of where this would be necessary we'd be past that point any way... :)

somik123, 2020-08-13:
Instead of editing the index.html for every addon, maybe it would be better to create a "addon system" where the data will be imported or redirect the user to/from another page? That way it would survive an update.

danmed, 2020-08-13:
So i had a thought.. i can just create another html page that combines the normal index.html and my power commands.. but when i try to hit the url (index2.html) i get a 404 error..

im guessing there's some nginx redirection at play here that i'm missing..

mtlynch, 2020-08-13:
I'd be interested in adding a plugin system, but it's not something I'd be able to add in the short-term. TinyPilot is still very new, so it's hard for me to design a plugin system before the core functionality has matured a bit.

I think the best short-term solution would be to create a fork that adds the buttons you need to the UI. The Ansible role supports configurable git repos and branches, so your fork should only have to change this line:

https://github.com/mtlynch/tinypilot/blob/ebf22575721ca984bbcf0ca1f27dac035dbda682/quick-install#L37

to:

```bash
ansible-playbook -i localhost, install.yml --extra-vars "tinypilot_repo=https://github.com/danmed/tinypilot.git tinypilot_repo_branch=smartplug"
```

Where `smartplug` is whatever you decide to name your branch. You'd still be able to pick up updates from TinyPilot by syncing your fork against the mainline repo.

danmed, 2020-08-13:
thank you.. i've been using github for a while but it still perplexes me as to how it all works.. i'll re-red this in the morning and see if i can figure it out!

alternatively.. ill host a page on my local webserver that combines tinypilot and my extra links..

it's sensible to focus on core functionality before all this extra stuff :)

somik123, 2020-08-15:
> So i had a thought.. i can just create another html page that combines the normal index.html and my power commands.. but when i try to hit the url (index2.html) i get a 404 error..
> 
> im guessing there's some nginx redirection at play here that i'm missing..

Are you sure you are placing it in the correct directory? This is where your mods.html should be located:
`/opt/tinypilot/app/static/mods.html`

lensherm, 2021-03-12:
Technically, if you add a simple MQTT broker, for example Eclipse Mosquitto, which happily runs on a Pi, you can interface with anything which talks MQTT, such as Tasmota smart plugs, or Shelly switches, etc. Looks like there's some work, which has been done to control TP-Link plugs using MQTT as well.