tiny-pilot/tinypilot #130: Can't run tiny pilot after recent update

ubenmackin, 2020-08-14:
Tinypilot used to run fine for me. I recently just re ran the install script:

```
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

But now when I navigate to the webpage, I get a blank white screen with the text "Not found" displayed. Checking systemctl, I can see tinypilot and ustreamer are running. Nginx is as well. Not sure why it stopped working after this install.

mtlynch, 2020-08-14:
Sorry you're running into this. I haven't run into that and am not able to reproduce it on my side. Can you try checking some of the logs?

```bash
sudo journalctl -u tinypilot
sudo journalctl -u nginx
sudo tail /var/log/nginx/error.log
sudo tail /var/log/nginx/access.log
```

ubenmackin, 2020-08-14:
Here are the logs.

```
pi@node2:~ $ sudo journalctl -u tinypilot
-- Logs begin at Thu 2019-02-14 02:12:00 PST, end at Fri 2020-08-14 15:08:24 PDT. --
Aug 14 12:58:26 node2 systemd[1]: Started TinyPilot - RPi-based virtual KVM.
Aug 14 12:58:33 node2 python[322]: [2020-08-14 12:58:33,633] INFO in main: Starting app
Aug 14 12:58:33 node2 python[322]: [2020-08-14 12:58:33,753] INFO in server: Server initialized for eventlet.
Aug 14 12:58:33 node2 python[322]: [2020-08-14 12:58:33,808] INFO in _internal:  * Restarting with stat
Aug 14 12:58:37 node2 python[322]: [2020-08-14 12:58:37,992] INFO in main: Starting app
Aug 14 12:58:38 node2 python[322]: [2020-08-14 12:58:38,091] INFO in server: Server initialized for eventlet.
```

```
pi@node2:~ $ sudo journalctl -u nginx
-- Logs begin at Thu 2019-02-14 02:12:00 PST, end at Fri 2020-08-14 15:08:51 PDT. --
Aug 14 12:58:26 node2 systemd[1]: Starting A high performance web server and a reverse proxy server...
Aug 14 12:58:27 node2 systemd[1]: Started A high performance web server and a reverse proxy server.
Aug 14 13:03:01 node2 systemd[1]: Stopping A high performance web server and a reverse proxy server...
Aug 14 13:03:01 node2 systemd[1]: nginx.service: Succeeded.
Aug 14 13:03:01 node2 systemd[1]: Stopped A high performance web server and a reverse proxy server.
Aug 14 13:03:01 node2 systemd[1]: Starting A high performance web server and a reverse proxy server...
Aug 14 13:03:01 node2 systemd[1]: Started A high performance web server and a reverse proxy server.
Aug 14 13:20:57 node2 systemd[1]: Reloading A high performance web server and a reverse proxy server.
Aug 14 13:20:57 node2 systemd[1]: Reloaded A high performance web server and a reverse proxy server.
```

```
pi@node2:~ $ sudo tail /var/log/nginx/error.log
2020/08/14 13:20:57 [notice] 4269#4269: signal process started
2020/08/14 13:21:03 [error] 4272#4272: *26297 connect() failed (111: Connection refused) while connecting to upstream, client: 10.100.1.42, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "node2.int.volcrypt.com"
2020/08/14 13:21:05 [error] 4272#4272: *26297 connect() failed (111: Connection refused) while connecting to upstream, client: 10.100.1.42, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "node2.int.volcrypt.com"
2020/08/14 13:21:27 [error] 4272#4272: *26297 connect() failed (111: Connection refused) while connecting to upstream, client: 10.100.1.42, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "node2.int.volcrypt.com"
2020/08/14 13:24:06 [error] 4272#4272: *26308 connect() failed (111: Connection refused) while connecting to upstream, client: 10.100.1.42, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "node2.int.volcrypt.com"
2020/08/14 13:24:18 [error] 4272#4272: *26307 connect() failed (111: Connection refused) while connecting to upstream, client: 10.100.1.42, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "node2.int.volcrypt.com"
```

```
pi@node2:~ $ sudo tail /var/log/nginx/access.log
10.100.1.42 - - [14/Aug/2020:13:21:03 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:21:04 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:21:05 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:21:26 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:21:27 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:22:13 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:24:06 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:24:12 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:24:18 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [14/Aug/2020:13:24:18 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
```

It looks like it is trying to connect to port 8000 locally, but it is failing?

I see something listening on 8000:

```
pi@node2:~ $ sudo netstat -tulpn | grep LISTEN | grep 8000
tcp        0      0 127.0.0.1:8000          0.0.0.0:*               LISTEN      382/python
```

But maybe it is an ipv6 issue? It looks like it is trying to connect to port 8000 via IPV6 (http://[::1]:8000/) but the python app is only listening on ipv4?

ubenmackin, 2020-08-14:
So I think it was an ipv6 vs ipv4 issue.

I changed the following in nginx from:

```
    upstream tinypilot {
        server localhost:8000 fail_timeout=2s max_fails=600;
    }
    upstream ustreamer {
        server localhost:8001 fail_timeout=2s max_fails=600;
    }
```

to

```
    upstream tinypilot {
        server 127.0.0.1:8000 fail_timeout=2s max_fails=600;
    }
    upstream ustreamer {
        server 127.0.0.1:8001 fail_timeout=2s max_fails=600;
    }
```

and I rebooted, and then it worked fine.

ubenmackin, 2020-08-15:
Hmm...well it just stopped working on me after working for a bit. I see this in the log:

`2020/08/14 15:23:42 [warn] 351#351: *5 an upstream response is buffered to a temporary file /var/lib/nginx/proxy/1/00/0000000001 while reading upstream, client: 10.100.1.42, server: tinypilot, request: "GET /stream?advance_headers=1 HTTP/1.1", upstream: "http://127.0.0.1:8001/stream?advance_headers=1", host: "node2.int.volcrypt.com", referrer: "http://node2.int.volcrypt.com/"`

Rebooting the pi brings things back up.

mtlynch, 2020-08-15:
When it doesn't work, is it just showing "Not found" or is it loading TinyPilot but failing to load the video feed?

ubenmackin, 2020-08-15:
> When it doesn't work, is it just showing "Not found" or is it loading TinyPilot but failing to load the video feed?

It just says Not found. It doesn’t load the TinyPilot webpage or interface at all.

It seems to run for about 5-10 minutes (sorry don’t have an exact time) after a reboot. Then it does.

I tried restarting all the services (ustream, tinypilot, ngoni) but just restarting those doesn’t bring things back. Have to actually reboot the whole pi.

somik123, 2020-08-15:
@ubenmackin 

When you get the Not Found error, can you please run these 2 commands and post the output?

```
sudo systemctl restart nginx
sudo cat /var/log/nginx/access.log
```

The first line restarts nginx and dumps access logs to the access.log file.
Second line prints the access log.

ubenmackin, 2020-08-15:
I think I stumbled on to something.

I am currently running this on a pi2b, because I wanted to test out the video piece while my pi4 was on order. The pi2 obviously can't do OTG, but I figured the keyboard part would just not work and the rest would.

It looks like there must be a check at some point for the keyboard/OTG interface after a period of time. I left up the site, and after awhile I got a big pink box that said something about the keyboard, but I refreshed the page (to see if it would go away) before I copied the error. After the refresh, I got the Not found error. I've tried repeating but can't get the pink box to appear again.

So do you know if there is a check for the OTG/keyboard, and if it is "Not found" the page would just error out?

Here are the logs:

```
pi@node2:~ $ sudo systemctl restart nginx
pi@node2:~ $ sudo cat /var/log/nginx/access.log
10.100.1.42 - - [15/Aug/2020:09:59:06 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:09:59:07 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:06 -0700] "GET /third-party/socket.io/2.3.0/socket.io.js HTTP/1.1" 200 68686 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:09:59:35 -0700] "GET / HTTP/1.1" 502 173 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:06 -0700] "GET / HTTP/1.1" 200 4763 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:06 -0700] "GET /css/style.css HTTP/1.1" 200 4605 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:06 -0700] "GET /js/app.js HTTP/1.1" 200 8997 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:07 -0700] "GET /socket.io/?EIO=3&transport=polling&t=NFp46x_ HTTP/1.1" 200 119 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:07 -0700] "GET /socket.io/?EIO=3&transport=polling&t=NFp46za&sid=f57efd8fe644452991008efff2a02a9e HTTP/1.1" 200 4 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:23 -0700] "GET /socket.io/?EIO=3&transport=websocket&sid=f57efd8fe644452991008efff2a02a9e HTTP/1.1" 101 8 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:24 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:06 -0700] "GET /img/GitHub-Mark-Light-120px-plus.png HTTP/1.1" 200 4044 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:06 -0700] "GET /img/power-icon.png HTTP/1.1" 200 4205 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:07 -0700] "GET /img/question-icon.svg HTTP/1.1" 200 1044 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:00:24 -0700] "GET /stream?advance_headers=1 HTTP/1.1" 200 56392805 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:01:06 -0700] "GET / HTTP/1.1" 200 4763 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:01:07 -0700] "GET /socket.io/?EIO=3&transport=polling&t=NFp4OL4 HTTP/1.1" 200 119 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:01:07 -0700] "GET /socket.io/?EIO=3&transport=polling&t=NFp4OMi&sid=13c7c1d2a69f4ee586adfa3ce5892f7b HTTP/1.1" 200 4 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:01:09 -0700] "GET /stream?advance_headers=1 HTTP/1.1" 200 5702789 "http://node2.int.volcrypt.com/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
[... cut ...] o) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:03:18 -0700] "GET / HTTP/1.1" 404 9 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
10.100.1.42 - - [15/Aug/2020:10:03:18 -0700] "GET /socket.io/?EIO=3&transport=websocket&sid=af904ff402fb45068cc009c740fad967 HTTP/1.1" 101 63 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15" "-"
```

somik123, 2020-08-15:
So you suspect that tinypilot is crashing when it cant connect to the HID device?

This line does suggest it should crash after about 10 mins (600 tries):
`server 127.0.0.1:8000 fail_timeout=1s max_fails=600;`

You can monitor if tinypilot or ustreamer crashed with the commands: 

```
sudo journalctl -u tinypilot
sudo journalctl -u ustreamer
```

Run those immediately after you get the 404 error.

mtlynch, 2020-10-30:
@ubenmackin - Did switching to the Pi 4 resolve this?

Sidenote: I added a convenience script for collecting and sharing logs:

```bash
/opt/tinypilot/dev-scripts/dump-logs
```

ubenmackin, 2020-10-31:
Sorry, yes it is working for me now on a rpi4.

mtlynch, 2020-10-31:
Glad to hear it!

closed 2020-10-31 by mtlynch