tiny-pilot/tinypilot #131: Configure WiFi options from the UI

mtlynch, 2020-08-14:
TinyPilot users can currently [configure their devices for WiFi via the command-line](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md), but it would be good if we offered a path to do it right from the web UI.

mtlynch, 2023-10-31:
Bumping priority as I recently received a request to enable this.

shalver-tp, 2024-05-21:
This will likely be a popup that allows the user to input the needed information.

shalver-tp, 2024-06-07:
@jotaen4tinypilot  That's a good question!

Let's make it a Community feature as I think a UI change like this makes sense to go across both versions. It isn't really a new "feature" that we would want to wall off.

jdeanwallace, 2024-06-12:
@jotaen4tinypilot - RE:

> Find most suitable way to access and alter the WiFi configuration programmatically

We have 2 existing scripts that enable/disable the WiFi access point, namely:
* [`/opt/tinypilot-privileged/scripts/enable-wifi-ap`](https://github.com/tiny-pilot/tinypilot/blob/7a31d856a6c2e5d8da3ef9c7f067a11f1b7a7e77/debian-pkg/opt/tinypilot-privileged/scripts/enable-wifi-ap)
* [`/opt/tinypilot-privileged/scripts/disable-wifi-ap`](https://github.com/tiny-pilot/tinypilot/blob/7a31d856a6c2e5d8da3ef9c7f067a11f1b7a7e77/debian-pkg/opt/tinypilot-privileged/scripts/disable-wifi-ap)

From what I can see, they might not require a reboot (haven't tested).

> I’m not 100% certain whether configuring WiFi could have any potential side effects with other networking features. I’m thinking of the “Static IP” feature in particular, because I remember that there might have been some sort of implication there with regards to wireless connections.

Yes, currently enabling/disabling the WiFi using the above scripts nukes any Static IP config too:
https://github.com/tiny-pilot/tinypilot/blob/7a31d856a6c2e5d8da3ef9c7f067a11f1b7a7e77/debian-pkg/opt/tinypilot-privileged/scripts/enable-wifi-ap#L37-L39

shalver-tp, 2024-06-12:
We had a bit of discussion related to wifi and static ips recently here: https://github.com/tiny-pilot/tinypilot-pro/issues/1340

Yes, this is more about getting to a dual NIC setup (Wi-Fi and wired at this point)

jdeanwallace, 2024-06-12:
> Wouldn’t configuring the device to act as WiFi access point be something slightly different from this task, though?

Oh yes, you're right. I didn't register the "access point" part.

shalver-tp, 2024-06-19:
@cghague @db39 FYI about the fallback scenario upon first feature release. (see the recent comment)