tiny-pilot/tinypilot #132: USB virtual keyboard not working when TinyPilot plugged in after boot

dougrathbone, 2020-08-15:
I find that when i plug in the KVM to my machine *after* its already booted, the KVM's virtual keyboard is not detected. This is problematic when trying to plug the device in to troubleshoot an issue remotely (eg lack of network connectivity on a host).

Motherboard: ASUS P8B75-V

Steps to reproduce:
- Unplug TinyPilot USB-C to USB-A
- Start host machine
- Plug in TinyPilot
- note an inability to use keyboard.

This works fine when the device is plugged in from boot.

mtlynch, 2020-08-15:
Oh, I haven't seen that. Which OS does it occur on? If you plug a physical keyboard into the target machine after boot, does it recognize the keyboard?

dougrathbone, 2020-08-15:
This is on Proxmox/Debian.

Plugging In a physical works. Would love to help debug but not sure where
to start.

What’s a good way to capture the usb handshake etc on the host?

mtlynch, 2020-08-18:
I'm not aware of a good way to capture the handshake.

It might be useful to see the output of `lsusb` when it's working vs. when it's not. Also `dmesg` right after you try plugging in the TinyPilot.

closed 2020-09-04 by mtlynch