tiny-pilot/tinypilot #133: tinypilot on raspbian only shows error message

probutus, 2020-08-15:
Did everything exactly as advertised. ustreamer working, usb keyboard demon working. When connecting to raspberrypi/ I get this message in the browser window: (also when connecting to raspberrypi:8000)

{"errcode":"0","errmsg":"ok","UTC_offset":"-18000"}

When running tinypilot manually I get this error message:

[13:10:15] root@raspberrypi:/opt/tinypilot# HOST=raspberrypi PORT=8000 KEYBOARD_PATH=/dev/hidg0 /opt/tinypilot/venv/bin/python /opt/tinypilot/app/main.py 
[2020-08-15 13:10:19,329] INFO in main: Starting app
[2020-08-15 13:10:19,348] INFO in server: Server initialized for eventlet.
[2020-08-15 13:10:19,357] INFO in _internal:  * Restarting with stat
[2020-08-15 13:10:20,092] INFO in main: Starting app
[2020-08-15 13:10:20,111] INFO in server: Server initialized for eventlet.

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 425, in resolve
    use_network=use_network)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 380, in query
    return end()
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 359, in end
    raise result[1]
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 340, in step
    a = fun(*args, **kwargs)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/dns/resolver.py", line 1091, in query
    True)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/dns/resolver.py", line 1043, in resolve
    timeout = self._compute_timeout(start, lifetime)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/dns/resolver.py", line 950, in _compute_timeout
    raise Timeout(timeout=duration)
dns.exception.Timeout: The DNS operation timed out after 5.107467412948608 seconds

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/flask_socketio/__init__.py", line 559, in run_server
    addresses = eventlet.green.socket.getaddrinfo(host, port)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 517, in getaddrinfo
    qname, addrs = _getaddrinfo_lookup(host, family, flags)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 490, in _getaddrinfo_lookup
    raise err
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 479, in _getaddrinfo_lookup
    answer = resolve(host, qfamily, False, use_network=use_network)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 432, in resolve
    raise EAI_EAGAIN_ERROR
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 479, in _getaddrinfo_lookup
    answer = resolve(host, qfamily, False, use_network=use_network)
  File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 432, in resolve
    raise EAI_EAGAIN_ERROR
socket.gaierror: [Errno -3] Lookup timed out

somik123, 2020-08-15:
@probutus Unable to reproduce.

Could you be more specific on version of Rasbian or Raspberry Pi OS?

Did you run:

```
sudo apt update
sudo apt dist-upgrade
sudo reboot
```

before you ran

```
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

probutus, 2020-08-15:
yes: as you can see the system is up to date

[14:18:29] root@raspberrypi:/home/tux# lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
[14:18:56] root@raspberrypi:/home/tux# apt-get update
OK:1 http://archive.raspberrypi.org/debian buster InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                  
Ign:3 https://dl.bintray.com/openhab/apt-repo2 stable InRelease                                                                                 
OK:4 https://packages.grafana.com/oss/deb stable InRelease                                                                                      
Holen:5 https://dl.bintray.com/openhab/apt-repo2 stable Release [6.051 B]                                                                       
OK:6 https://repos.influxdata.com/debian buster InRelease                                                                     
OK:7 https://apt.homegear.eu/Raspbian buster/ InRelease
Es wurden 6.051 B in 2 s geholt (2.912 B/s).
Paketlisten werden gelesen... Fertig
[14:19:05] root@raspberrypi:/home/tux# apt-get dist-upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  rpi-eeprom-images
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

somik123, 2020-08-15:
When you run the install, the last line shows any fails or errors?

```
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - 
```

Yes, you can run the install script again even if you already have it installed.

probutus, 2020-08-15:
many thanks for your patience and your help! It's much appreciated!!
Here is the complete install log:
[14:21:47] tux@raspberrypi:~$ curl \
>   --silent \
>   --show-error \
>   https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
>     bash - && \
>   sudo reboot
+ set -e
+ set -u
+ '[' -f /etc/nginx/sites-enabled/KVMPi.conf ']'
++ mktemp -d
+ pushd /tmp/tmp.obQf4mJvHg
/tmp/tmp.obQf4mJvHg ~
+ sudo apt-get update
OK:1 http://archive.raspberrypi.org/debian buster InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                  
Ign:3 https://dl.bintray.com/openhab/apt-repo2 stable InRelease                                                                                 
OK:4 https://packages.grafana.com/oss/deb stable InRelease                                                                                      
Holen:5 https://dl.bintray.com/openhab/apt-repo2 stable Release [6.051 B]                                                     
OK:6 https://apt.homegear.eu/Raspbian buster/ InRelease                                                                               
OK:7 https://repos.influxdata.com/debian buster InRelease                                             
Es wurden 6.051 B in 2 s geholt (2.940 B/s).
Paketlisten werden gelesen... Fertig
+ sudo apt-get install -y libssl-dev python3-dev python3-venv
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
python3-dev ist schon die neueste Version (3.7.3-1).
python3-venv ist schon die neueste Version (3.7.3-1).
libssl-dev ist schon die neueste Version (1.1.1d-0+deb10u3+rpt1).
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  rpi-eeprom-images
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
+ python3 -m venv venv
+ . venv/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/tmp/tmp.obQf4mJvHg/venv
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ PATH=/tmp/tmp.obQf4mJvHg/venv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ export PATH
++ '[' -n '' ']'
++ '[' -z '' ']'
++ _OLD_VIRTUAL_PS1=
++ '[' 'x(venv) ' '!=' x ']'
++ PS1='(venv) '
++ export PS1
++ '[' -n /bin/bash -o -n '' ']'
++ hash -r
+ pip install wheel==0.34.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting wheel==0.34.2
  Downloading https://files.pythonhosted.org/packages/8c/23/848298cccf8e40f5bbb59009b32848a4c38f4e7f3364297ab3c3e2e2cd14/wheel-0.34.2-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.34.2
+ pip install ansible==2.9.10
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible==2.9.10
  Downloading https://files.pythonhosted.org/packages/03/4f/cccab1ec2e0ecb05120184088e00404b38854809cf35aa76889406fbcbad/ansible-2.9.10.tar.gz (14.2MB)
    100% |████████████████████████████████| 14.2MB 31kB/s 
Collecting PyYAML (from ansible==2.9.10)
[... 183 lines cut ...]
ok: [localhost]

TASK [mtlynch.tinypilot : save absolute path to python3] ***********************
ok: [localhost]

TASK [mtlynch.tinypilot : create TinyPilot virtualenv] *************************
ok: [localhost]

TASK [mtlynch.tinypilot : fix TinyPilot folder permissions] ********************
ok: [localhost]

TASK [mtlynch.tinypilot : install TinyPilot as a service] **********************
ok: [localhost]

TASK [mtlynch.tinypilot : enable systemd TinyPilot service file] ***************
ok: [localhost]

RUNNING HANDLER [geerlingguy.nginx : reload nginx] *****************************
changed: [localhost]

PLAY RECAP *********************************************************************
localhost                  : ok=41   changed=2    unreachable=0    failed=0    skipped=9    rescued=0    ignored=0   

Connection to raspberrypi closed by remote host.
Connection to raspberrypi closed.

somik123, 2020-08-15:
Once rebooted, try to access the tinypilot by entering your raspberry pi IP into your browser. Everything should run with boot so you should not need to start up anything on your own.

probutus, 2020-08-15:
I did; same result independently if I use the ipv4 or ipv6 or the hostname. I always get this output:

{"errcode":"0","errmsg":"ok","UTC_offset":"-18000"}

somik123, 2020-08-15:
> I did; same result independently if I use the ipv4 or ipv6 or the hostname. I always get this output:
> 
> {"errcode":"0","errmsg":"ok","UTC_offset":"-18000"}

You are connecting on port 80 right? Like: http://192.168.1.123/

probutus, 2020-08-15:
yes
also http://192.168.1.186:80 is the same result

somik123, 2020-08-15:
Whats the output of:
`ls /etc/nginx/sites-enabled/`

and

```
sudo journalctl -u tinypilot
sudo journalctl -u nginx
sudo tail /var/log/nginx/error.log
sudo tail /var/log/nginx/access.log
```

probutus, 2020-08-15:
ls /etc/nginx/sites-enabled/
`[14:40:43] root@raspberrypi:/home/tux# ls /etc/nginx/sites-enabled/
tinypilot.conf  weewx.conf
`

sudo journalctl -u tinypilot
-- Reboot --
Aug 14 18:17:16 raspberrypi systemd[1]: Started TinyPilot - RPi-based virtual KVM.
Aug 14 18:17:25 raspberrypi python[621]: [2020-08-14 18:17:25,170] INFO in main: Starting app
Aug 14 18:17:25 raspberrypi python[621]: [2020-08-14 18:17:25,253] INFO in server: Server initialized for eventlet.
Aug 14 18:17:25 raspberrypi python[621]: [2020-08-14 18:17:25,274] INFO in _internal:  * Restarting with stat
Aug 14 18:17:26 raspberrypi python[621]: [2020-08-14 18:17:26,175] INFO in main: Starting app
Aug 14 18:17:26 raspberrypi python[621]: [2020-08-14 18:17:26,200] INFO in server: Server initialized for eventlet.
Aug 14 18:38:51 raspberrypi systemd[1]: Stopping TinyPilot - RPi-based virtual KVM...
Aug 14 18:38:51 raspberrypi systemd[1]: tinypilot.service: Succeeded.
Aug 14 18:38:51 raspberrypi systemd[1]: Stopped TinyPilot - RPi-based virtual KVM.
Aug 14 18:40:22 raspberrypi systemd[1]: Started TinyPilot - RPi-based virtual KVM.

sudo tail /var/log/nginx/error.log
[14:41:57] root@raspberrypi:/home/tux# sudo tail /var/log/nginx/error.log
2020/08/15 12:58:46 [error] 799#799: *21 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.1.167, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "raspberrypi"
2020/08/15 13:18:51 [error] 799#799: *26 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.1.167, server: tinypilot, request: "GET / HTTP/1.1", upstream: "http://[::1]:8000/", host: "raspberrypi"
2020/08/15 14:13:19 [notice] 9404#9404: using inherited sockets from "6;7;"
2020/08/15 14:25:16 [notice] 4803#4803: signal process started

[14:41:58] root@raspberrypi:/home/tux# sudo tail /var/log/nginx/access.log
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weekwinddir.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weekwindvec.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weekrain.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weekuv.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weekradiation.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weektempin.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:27:26 +0200] "GET /weektemp.png HTTP/1.1" 304 0 "http://raspberrypi:81/" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:30:33 +0200] "GET / HTTP/1.1" 200 51 "-" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:30:39 +0200] "GET / HTTP/1.1" 200 51 "-" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"
192.168.1.167 - - [15/Aug/2020:14:31:18 +0200] "GET / HTTP/1.1" 200 51 "-" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:79.0) Gecko/20100101 Firefox/79.0" "-"

probutus, 2020-08-15:
the weewx instance is running on port 81 ok; no difference ẃhen I disable it

somik123, 2020-08-15:
> the weewx instance is running on port 81 ok; no difference ẃhen I disable it

That should not interfere with it.

This seems to be the same IPv6 issue, but your latest update should have fixed it...

Can you please change the following in /etc/nginx/nginx.conf from:

```
    upstream tinypilot {
        server localhost:8000 fail_timeout=1s max_fails=600;
    }
    upstream ustreamer {
        server localhost:8001 fail_timeout=1s max_fails=600;
    }
```
to

```
    upstream tinypilot {
        server 127.0.0.1:8000 fail_timeout=1s max_fails=600;
    }
    upstream ustreamer {
        server 127.0.0.1:8001 fail_timeout=1s max_fails=600;
    }
```

probutus, 2020-08-15:
the file is already like that; i restarted nginx but have the same result
I also disabled ipv6 completely on the raspi temporary but had the same issue

    upstream tinypilot {
        server 127.0.0.1:8000 fail_timeout=1s max_fails=600;
    }
    upstream ustreamer {
        server 127.0.0.1:8001 fail_timeout=1s max_fails=600;
    }

mtlynch, 2020-08-18:
@probutus - What behavior are you currently seeing?

This fix should have addressed the IPv6 issue:

https://github.com/mtlynch/ansible-role-tinypilot/pull/38

If you reinstall, you should pick up the changes without having to manually change any `nginx` files anymore.

I notice that your Pi is configured for German. I've never tested on anything besides English. I'm not seeing anything in the logs that would point to language as the issue, but it's a potential avenue of investigation.

probutus, 2020-08-18:
@mtlynch : behaviour on the raspi is pretty much the same. 
I am using a raspbian system which has also openhab installed but based on the latest raspian install. I see that if I install tinypilot standalone on a standard raspbian without any other components everything works as advertised (even in the German environment). So, feel free to close this issue; nevertheless I would be very happy to get a hint where this dns error could come from (this saves me one raspi4 for other topics ;-) )

mtlynch, 2020-08-19:
@probutus - when you say the same, you mean you're still seeing this error when you run TinyPilot?

```text
Exception in thread Thread-1:
Traceback (most recent call last):
File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 425, in resolve
use_network=use_network)
File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 380, in query
return end()
File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 359, in end
raise result[1]
File "/opt/tinypilot/venv/lib/python3.7/site-packages/eventlet/support/greendns.py", line 340, in step
a = fun(*args, **kwargs)
File "/opt/tinypilot/venv/lib/python3.7/site-packages/dns/resolver.py", line 1091, in query
True)
File "/opt/tinypilot/venv/lib/python3.7/site-packages/dns/resolver.py", line 1043, in resolve
timeout = self._compute_timeout(start, lifetime)
File "/opt/tinypilot/venv/lib/python3.7/site-packages/dns/resolver.py", line 950, in _compute_timeout
raise Timeout(timeout=duration)
dns.exception.Timeout: The DNS operation timed out after 5.107467412948608 seconds
```

Is it possible something else on your system is changing your DNS settings?

If you check `cat /etc/resolv.conf` on a working system vs. the non-working system, do you see different settings?

probutus, 2020-08-19:
@mtlynch : I checked; resolv.conf is on both systems (working and non working) automatically generated:

# Generated by resolvconf
domain universe
nameserver 127.0.0.1

when I do resolvconf -l I get:
# resolv.conf from lo.dnsmasq
nameserver 127.0.0.1

# resolv.conf from eth0.dhcp
# Generated by dhcpcd from eth0.dhcp
domain universe
search universe
nameserver 192.168.1.2

# resolv.conf from eth0.dhcp6
# Generated by dhcpcd from eth0.dhcp6
nameserver fdc7:34f5:cb96::1

# resolv.conf from eth0.ra
# Generated by dhcpcd from eth0.ra
nameserver fdc7:34f5:cb96::1

Address resolution via nslookup is working immediately and without any delay for local and global addresses:

[23:01:58] tux@raspberrypi:~$ nslookup google.com
Server:         127.0.0.1
Address:        127.0.0.1#53

Non-authoritative answer:
Name:   google.com
Address: 172.217.16.206
Name:   google.com
Address: 2a00:1450:4001:817::200e

[23:02:13] tux@raspberrypi:~$ nslookup nas
Server:         127.0.0.1
Address:        127.0.0.1#53

Name:   nas.universe
Address: 192.168.1.5

[23:02:26] tux@raspberrypi:~$ 

Trying a non-existing address will also result in an immediate error message (no delay):
[23:04:47] tux@raspberrypi:~$ nslookup derwesfdsdfsdafsafasdfasfd.com
Server:         127.0.0.1
Address:        127.0.0.1#53

** server can't find derwesfdsdfsdafsafasdfasfd.com: NXDOMAIN

[23:04:55] tux@raspberrypi:~$ 

Is there a way to figure out which address is requested by tinypilot on startup that causes the dnslookup to time out?

mtlynch, 2020-08-19:
>Is there a way to figure out which address is requested by tinypilot on startup that causes the dnslookup to time out?

I'm not sure. I can't think of anything in TinyPilot that would even be resolving a hostname with DNS. Everything is local.

probutus, 2020-08-24:
I did now the install the other way around. Started with raspian, put on tinypilot and then added the other services. Now its working. Let's close this topic

closed 2020-08-24 by probutus