tiny-pilot/tinypilot #137: Error running quick_install to update

danmed, 2020-08-17:
I'm receiving this error when running quick_install to update the version : 

Command "/tmp/tmp.g8xVFkS0KG/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-7ajrzbrn/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-cvqkxub2/install-record.txt --single-version-externally-managed --compile --install-headers /tmp/tmp.g8xVFkS0KG/venv/include/site/python3.7/cffi" failed with error code 1 in /tmp/pip-install-7ajrzbrn/cffi/

This is on a pi4 using the recommended distro.

mtlynch, 2020-08-17:
Thanks for reporting this! You're seeing this repeatedly?

Could you share a [gist](https://gist.github.com/) of the full output?

danmed, 2020-08-17:
Yeah i'm seeing it repeatedly even after a reboot..

gimme a few mins and i'll post a gist

mtlynch, 2020-08-18:
Hmm, not sure why it's suddenly failing without this package. This should fix it:

```bash
sudo apt-get install -y libffi-dev
```

PR #138 (2020-08-18): Adding libffi-dev to list of ansible requirements
Fixes #137
closed 2020-08-18 by mtlynch