tiny-pilot/tinypilot #139: Keyboard and now screen capture issue: Exception in thread Thread-2

jfreak53, 2020-08-18:
I installed according to Wiki, everything went without any errors, I've even tried running the install again to try and re-install/upgrade the scripts to see if that would fix something.

The screen capture was working, it's not anymore, just shows black screen now:

[image]

I am accessing via the url `http://raspberrypi/`, shows green for keyboard. When it was capturing screen I'd press a key on the keyboard and nothing happened.

Here is my `journalctl -f` output:

```
Aug 18 17:05:35 raspberrypi systemd[574]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Aug 18 17:05:35 raspberrypi systemd[574]: Reached target Timers.
Aug 18 17:05:35 raspberrypi systemd[574]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Aug 18 17:05:35 raspberrypi systemd[574]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Aug 18 17:05:35 raspberrypi systemd[574]: Reached target Sockets.
Aug 18 17:05:35 raspberrypi systemd[574]: Reached target Basic System.
Aug 18 17:05:35 raspberrypi systemd[574]: Reached target Default.
Aug 18 17:05:35 raspberrypi systemd[574]: Startup finished in 145ms.
Aug 18 17:05:35 raspberrypi systemd[1]: Started User Manager for UID 0.
Aug 18 17:05:35 raspberrypi systemd[1]: Started Session c1 of user root.
Aug 18 17:05:40 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Aug 18 17:05:43 raspberrypi python[549]: [2020-08-18 17:05:43,491] INFO in socket: b33873a0a261445691ccd788244e525b: Received packet MESSAGE data 2["keystroke",{"metaKey":false,"altKey":false,"shiftKey":false,"ctrlKey":false,"key":"Enter","keyCode":13,"location":null}]
Aug 18 17:05:43 raspberrypi python[549]: [2020-08-18 17:05:43,492] INFO in server: received event "keystroke" from b33873a0a261445691ccd788244e525b [/]
Aug 18 17:05:43 raspberrypi kernel: configfs-gadget gadget: usb_ep_queue error on int endpoint -11
Aug 18 17:05:43 raspberrypi python[549]: Exception in thread Thread-2:
Aug 18 17:05:43 raspberrypi python[549]: Traceback (most recent call last):
Aug 18 17:05:43 raspberrypi python[549]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Aug 18 17:05:43 raspberrypi python[549]:     self.run()
Aug 18 17:05:43 raspberrypi python[549]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Aug 18 17:05:43 raspberrypi python[549]:     self._target(*self._args, **self._kwargs)
Aug 18 17:05:43 raspberrypi python[549]:   File "/opt/tinypilot/app/hid.py", line 23, in _write_to_hid_interface
Aug 18 17:05:43 raspberrypi python[549]:     hid_handle.write(bytearray(buffer))
Aug 18 17:05:43 raspberrypi python[549]: BlockingIOError: [Errno 11] write could not complete without blocking
Aug 18 17:05:43 raspberrypi python[549]: Exception in thread Thread-3:
Aug 18 17:05:43 raspberrypi python[549]: Traceback (most recent call last):
Aug 18 17:05:43 raspberrypi python[549]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Aug 18 17:05:43 raspberrypi python[549]:     self.run()
Aug 18 17:05:43 raspberrypi python[549]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Aug 18 17:05:43 raspberrypi python[549]:     self._target(*self._args, **self._kwargs)
Aug 18 17:05:43 raspberrypi python[549]:   File "/opt/tinypilot/app/hid.py", line 23, in _write_to_hid_interface
Aug 18 17:05:43 raspberrypi python[549]:     hid_handle.write(bytearray(buffer))
Aug 18 17:05:43 raspberrypi python[549]: BlockingIOError: [Errno 11] write could not complete without blocking
Aug 18 17:05:43 raspberrypi python[549]: [2020-08-18 17:05:43,507] INFO in server: emitting event "keystroke-received" to all [/]
Aug 18 17:05:43 raspberrypi kernel: configfs-gadget gadget: usb_ep_queue error on int endpoint -11
Aug 18 17:05:43 raspberrypi python[549]: [2020-08-18 17:05:43,509] INFO in socket: b33873a0a261445691ccd788244e525b: Client is gone, closing socket
Aug 18 17:05:43 raspberrypi python[549]: [2020-08-18 17:05:43,512] INFO in main: Client disconnected
Aug 18 17:05:43 raspberrypi python[549]: [2020-08-18 17:05:43,512] INFO in socket: b33873a0a261445691ccd788244e525b: Client is gone, closing socket
Aug 18 17:05:46 raspberrypi python[549]: [2020-08-18 17:05:46,461] INFO in socket: b33873a0a261445691ccd788244e525b: Received packet PING data None
Aug 18 17:05:46 raspberrypi python[549]: [2020-08-18 17:05:46,462] INFO in socket: Receive error -- socket is closed
Aug 18 17:05:47 raspberrypi python[549]: [2020-08-18 17:05:47,964] INFO in socket: 7682a4ae0d484c42b92173db802998ce: Sending packet OPEN data {'sid': '7682a4ae0d484c42b92173db802998ce', 'upgrades': ['websocket'], 'pingTimeout': 60000, 'pingInterval': 25000}
Aug 18 17:05:47 raspberrypi python[549]: [2020-08-18 17:05:47,967] INFO in main: Client connected
Aug 18 17:05:47 raspberrypi python[549]: [2020-08-18 17:05:47,968] INFO in socket: 7682a4ae0d484c42b92173db802998ce: Sending packet MESSAGE data 0
Aug 18 17:05:47 raspberrypi python[549]: [2020-08-18 17:05:47,985] INFO in socket: 7682a4ae0d484c42b92173db802998ce: Received request to upgrade to websocket
Aug 18 17:05:48 raspberrypi python[549]: [2020-08-18 17:05:48,006] INFO in socket: 7682a4ae0d484c42b92173db802998ce: Upgrade to websocket successful
Aug 18 17:06:13 raspberrypi python[549]: [2020-08-18 17:06:13,084] INFO in socket: 7682a4ae0d484c42b92173db802998ce: Received packet PING data None
Aug 18 17:06:13 raspberrypi python[549]: [2020-08-18 17:06:13,084] INFO in socket: 7682a4ae0d484c42b92173db802998ce: Sending packet PONG data None
```

I am running it on `Raspbian GNU/Linux 10 (buster)` Lite download, fresh image/install, only change was Wifi and SSH install before boot.

```
# uname -a
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

Not quite sure what's going on or not working here. Any thoughts?

mtlynch, 2020-08-18:
Sorry for the issue. Thanks for reporting it!

Which model Raspberry Pi are you using? Can you share the model of USB cable you're using to connect the Pi to the computer?

Lastly, can you SSH in and try this command and tell me what happens?

```bash
echo -ne "\x20\0\xb\0\0\0\0\0" > /dev/hidg0 && \
  echo -ne "\0\0\0\0\0\0\0\0" > /dev/hidg0
```

jfreak53, 2020-08-18:
No prob! Bugs happen and it's probably some kink on this end we need to figure a way around. So interesting, same cables work for another PC, exact same PC, only difference is it has HDMI #1, and #2 it has an OS.

What I noticed was, it still would not accept keyboard commands "before" the OS loaded. I had it plugged into USB 2 on each machine.

I ran the command on the offending PC, this is what it gives me back:

`-bash: echo: write error: Resource temporarily unavailable`

On the PC that nothing is working anymore, I have DVI and VGA available, so I've tried both HDMI to VGA and HDMI to DVI, neither are working. The HDMI to DVI did work for about 10 minutes originally, once I rebooted it no longer worked.

This is the PI: https://www.amazon.com/gp/product/B07TD43PDZ

Would teamviewer be helpful? I can get you a teamviewer session into the PI to have a look through it.

mtlynch, 2020-08-18:
Are you saying it's fully functional when connected to one PC, but it doesn't work when connected to another PC? Or it's all the same target PC and it works when the OS is loaded but not in pre-boot?

>This is the PI: https://www.amazon.com/gp/product/B07TD43PDZ

That's the Pi 4B, so that should be compatible.

>I ran the command on the offending PC, this is what it gives me back:

>`-bash: echo: write error: Resource temporarily unavailable`

Sorry, I meant to run that command on the Pi. Or did you mean you ran it on the Pi while *connected* to the bad PC?

Can you share details about the working/non-working target computer(s)? Their OS and motherboard models would be helpful.

jfreak53, 2020-08-18:
> Are you saying it's fully functional when connected to one PC, but it doesn't work when connected to another PC? Or it's all the same target PC and it works when the OS is loaded but not in pre-boot?

Sorry, fully functional when connected to PC B, with OS after boot only, NOT functional on PC B before boot/during post. Not functional at all when connected to PC A.

PC B = Ubuntu OS, has HDMI

PC A = No OS, has DVI and VGA

> Sorry, I meant to run that command on the Pi. Or did you mean you ran it on the Pi while _connected_ to the bad PC?

Correct, I ran it on the Pi via SSH when it was connected to PC A.

> Can you share details about the working/non-working target computer(s)? Their OS and motherboard models would be helpful.

OS for PC B is Ubuntu, both PC's are HP P6140F's.

jfreak53, 2020-08-18:
I can confirm, keyboard works on PC A, after Post. If I boot into a Windows 10 boot drive, into the installer, keyboard now works. So keyboard works after the OS loads on PC A.

Still can't get Video to work though.

jfreak53, 2020-08-18:
I can also confirm that DVI to HDMI works on PC B, same cables. VGA to HDMI cannot get to work, which I can deal with, later I'll try different adapter down the road.

jfreak53, 2020-08-18:
Got video back on PC A now, something with moving cables wasn't working good. I'll figure out VGA adapters later. My only issue is then that keyboard doesn't work pre-OS loading on any PC I try. Same errors as above pre-os, once any OS loads, even boot DVD, it works fine.

somik123, 2020-08-18:
> I can also confirm that DVI to HDMI works on PC B, same cables. VGA to HDMI cannot get to work, which I can deal with, later I'll try different adapter down the road.

When you say vga to hdmi, do you mean hdmi to vga converter or vga to hdmi upscaler? I'm asking because hdmi to vga goes only one way. Hdmi output to vga input.

I had issues with lower end cables not working. When I switched to my phone's USB C charge and data cable, it worked fine. Could you maybe try something similar?

mtlynch, 2020-08-19:
@jfreak53 - For one of my machines, it won't accept keyboard immediately after boot. Any key I enter shows up red in TinyPilot. After 1-2 seconds, it begins accepting keystrokes, and they show up green. So if I hold down the key, it eventually picks it up pre-OS. Is it possible that your machine has similar behavior?

jfreak53, 2020-08-19:
> When you say vga to hdmi, do you mean hdmi to vga converter or vga to hdmi upscaler? I'm asking because hdmi to vga goes only one way. Hdmi output to vga input.

Ah, maybe that's it, I'll have a look.

> I had issues with lower end cables not working. When I switched to my phone's USB C charge and data cable, it worked fine. Could you maybe try something similar?

I will test it.

> For one of my machines, it won't accept keyboard immediately after boot. Any key I enter shows up red in TinyPilot. After 1-2 seconds, it begins accepting keystrokes, and they show up green. So if I [...]

I've left it on for a good 30 min on the boot screen before, then hit the keys, same thing. I never get red in the logs for tinypilot, just green when I hit a key.

closed 2021-03-09 by mtlynch