tiny-pilot/tinypilot #142: Blank screen when entering Intel NUC BIOS

rca, 2020-08-19:
The Intel NUC has a "Visual BIOS", which is pretty nice; definitely the prettiest BIOS I've seen.  It not only has keyboard input, but you can also point and click with the mouse.  It's definitely a pleasant user experience.

[image]

However, this is why I think there's trouble with Tiny Pilot.

On power up, the Intel NUC splash screen shows up.  After I hit `F2` to enter the BIOS and the splash screen goes away, the BIOS screen never shows up.  This may be related to #104, but it also seems different because it's consistently blank for the BIOS, but consistently fine otherwise.  When I hit `ctrl+alt+delete` at the blank screen, the machine will restart, the Intel NUC splash screen appears and the normal boot sequence displays just fine.

While debugging I noticed this interesting bit: If I plug in a monitor (an older 1080p Dell) and enter the BIOS, then plug Tiny Pilot back in, the BIOS does show up and I can interact with it.  Might there be some sort of HDMI compatibility checks that are being made by the BIOS prior to displaying; perhaps a setting that the monitor has, but the conversion card does not?

Thank you.

mtlynch, 2020-08-19:
Interesting, I'll try to get a hold of a NUC to debug this.

What model are you seeing this on?

Teddy-Schmitz, 2020-11-06:
I can report the same issue with my system.  It's an old Gigabyte H87N-Wifi.  I get a blank screen for the BIOS, but if I plug it into a monitor then back into the Tiny Pilot it displays just fine.  Let me know how I can help debug this.

mtlynch, 2020-11-10:
@Teddy-Schmitz - Oh, that's interesting behavior. If you plug TinyPilot in during BIOS, it doesn't capture the output, but if you plug it in after plugging and disconnecting a monitor, it behaves correctly?

Teddy-Schmitz, 2020-11-11:
Yes thats exactly it, just like the NUC its a UEFI Bios and I'm using the on-board video.

Teddy-Schmitz, 2020-11-12:
I did some more testing, if I boot into Linux (with no graphical interface) the screen stays blank even after plugging it into the monitor.

If I stay in the BIOS it works fine, if I do a soft reset it will continue to work until it starts the boot process into linux.

I'm running a Core i7-4770s.

mtlynch, 2020-11-13:
I can't think of why that would be. I'm planning to pick up a NUC to be able to debug locally, but I haven't had bandwidth. Some places you might check the logs.

This is a bit of a shot in the dark, but could you try restarting TinyPilot during BIOS, then entering these commands to see if there are any clues in the logs?

```bash
dmesg
```

```bash
sudo v4l2-ctl --all
```

```bash
sudo journalctl -u ustreamer | tail -n 200
```

Teddy-Schmitz, 2020-11-14:
Some more news, if I unplug the HDMI capture stick and plug it back in.  The screen shows up, but I have to do this everytime the system reboots.

I'll try to get you those logs soon.

mtlynch, 2021-03-09:
I believe this is a problem with the TinyPilot EDID (the blob that the HDMI capture passes to the target system to negotiate resolution and video capabilities). I've heard reports from NUC users that adding the following lines to `/home/tinypilot/settings.yml` fixes the issue:

```yaml
# EDID compatible with Intel NUC computers
ustreamer_edid: |
  00ffffffffffff0052628888008888881c150103800000780aee91a3544c
  99260f505425400001000100010001000100010001010101d51b00505000
  19400820b80080001000001eec2c80a070381a403020350040442100001e
  000000fc0050492d4b564d20566964656f0a000000fd00323d0f2e0f0000
  000000000000014d02030400de0d20a03058122030203400f0b400000018
  e01500a04000163030203400000000000018b41400a050d0112030203500
  80d810000018ab22a0a050841a3030203600b00e11000018000000000000
  000000000000000000000000000000000000000000000000000000000000
  00000000000000000000000000000045
```

After updating `settings.yml`, run `/opt/tinypilot/scripts/upgrade` to apply the settings.

closed 2021-03-09 by mtlynch

jhh, 2021-07-28:
If you arrived here by searching for this problem, I had to first upgrade my NUC (NUC7i3BNB) to a more recent version for the EDID fix mentioned [above](https://github.com/tiny-pilot/tinypilot/issues/142#issuecomment-794416779) to work.

I upgraded from a 2017 to a 2020 BIOS and can now view and operate the Intel Visual BIOS just fine.