tiny-pilot/tinypilot #173: Cannot change listening port after install

MikeRotch76, 2020-08-28:
I initially installed Tinypilot using the [https://github.com/mtlynch/tinypilot#simple-installation](url).

It worked fine.  However, I am now trying to change the listen port to something else.
I tried

_export TINYPILOT_INSTALL_VARS="tinypilot_port: 44556"
curl --silent --show-error https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | bash - &&_

and then rebooting, but listening port did not change
I confirmed during the install that the install var was being read by the installer:

_+ echo 'Install vars: tinypilot_port: 44556'
Install vars: tinypilot_port: 44556_

Is there a way I can fully uninstall TinyPilot or wipe the settings so I can do a clean install?

mtlynch, 2020-08-28:
Thanks for filing this!

TinyPilot actually listens on two separate ports. By default it listens for external connections on port 80 (via nginx) and then nginx forwards messages directly to the app, listening at 8000.

I've clarified this distinction and made it easier to specify a port other than 80:

https://github.com/mtlynch/ansible-role-tinypilot/pull/46/files

To listen for external connections on port 44556, you can do this:

```bash
export TINYPILOT_INSTALL_VARS="tinypilot_external_port=44556"
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

>Is there a way I can fully uninstall TinyPilot or wipe the settings so I can do a clean install?

In general, you should never have to uninstall TinyPilot to fix settings. Every time you install, the installer makes sure that anything you configured in a previous installation is updated with whatever your new settings are.

closed 2020-08-28 by mtlynch

MikeRotch76, 2020-08-30:
Thanks.  This works perfectly.

MikeRotch76, 2020-08-31:
Thanks Michael.  Thanks for resolving this so quickly and also adding the
M, in KVM.  I am eagerly awaiting the update to the power issue although
powering from a USB 3.0 port seems to be working fine (as you've stated in
your post).  But of course, I would like to meet the 3A spec.  I donated
$20 to you as a thank you for your hard work.

Transaction ID 1908289K4308231M