tiny-pilot/tinypilot #176: Power: Supplying 3A over USB-C

sstur, 2020-08-29:
Just opening this to track the power issue described in the [v2 blog post](https://tinypilotkvm.com/blog/v2-design).

The previously attempted solution was to use a [USB-C OTG Y-cable](https://tinypilotkvm.com/assets/static/y-cable.2671d65.bf14ad9dfe719710c1f17d9fda9eb4fb.jpg) but has this problem (quoted from blog post):

> A reader reached out to inform me that these cables do not support external power in the way I intended to use them for TinyPilot and risked damage to a user's USB ports. I paused sales and worked with an electrical engineering firm to confirm this, and I'm working on another solution. I've reached out to all v2 customers to alert them to this issue and provide updated parts once they're available.

I'd love to hear more about the other solution you're working on. In the mean time, I have a few suggestions that come to mind, which I'll try to outline below (and I can test at least one of these).

1. [UGREEN USB C Hub 5 in 1 Type C 3.1](https://www.amazon.com/gp/product/B077FZQS95)
[image]

2. [IOGEAR USB-C 4-Port Hub w/ 100W Power Delivery 3.0 Pass-Through](https://www.amazon.com/gp/product/B07D1GHW85)
[image]

3. [Targus USB-C Demultiplexer](https://www.amazon.com/gp/product/B074M2VTH4)
[image]

---

The first two are hubs, and there are probably dozens of variations of this concept, in which you can pass through USB-C power and plug in USB-A devices. I am skeptical this will work because it seems too obvious of a solution, but I have ordered the first one above and I'll try it on my setup when I can.

The third one seems more promising to me, and it's an unusual adapter that I stumbled across. Targus sells a multiplexer and demultiplexer variation, which in my mind means it splits/joins power and data for USB-C, although I have not been able to test this yet.

Even if one of those above works, I realize it's not a simple cable-based solution and they are each more expensive than a Pi, so that's not ideal.

But I'd love to hear more suggestions of how this might be solved.

mtlynch, 2020-08-29:
I don't think the hubs will work because they need to be able to provide the USB OTG functionality in addition to delivering power. The USB-C demultiplexer seems like it's the reverse of what we'd need in that it splits a source of power and USB data, when we actually want to take power in.

I'm working on a PCB that will split the power line from the USB output. It should be ready to ship from the US in a few weeks.

I've tested [USB/PWR Splitter](https://www.tindie.com/products/8086net/usbpwr-splitter/) successfully with the TinyPilot, and it's wired similarly to how TinyPilot's PCB will work. There's also [power blough-r](https://hackaday.io/project/160848-power-blough-r), which you could put in front of a Y-cable to protect the target computer from power backflow.

sstur, 2020-08-29:
> I'm working on a PCB that will split the power line from the USB output.

Wow, that's next level, this is amazing!

> The USB-C demultiplexer seems like it's the reverse of what we'd need

Got it, I think you're right (and the reverse product seems to be [this](https://www.amazon.com/gp/product/B074M2VTH4), although it's made for a vendor-specific docking station, and it's expensive).

Are you going down the path of USB-C PD to deliver 3A of power, or is that within the specs of USB-A?

If this can deliver PD 18+w, I can already think of other cool uses for this (fast charging a phone while plugged into USB-A CarPlay, among others).

Anyway, this is awesome, thanks for the details!

mtlynch, 2020-08-30:
>Are you going down the path of USB-C PD to deliver 3A of power, or is that within the specs of USB-A?

No, we don't need to do anything fancy to deliver 3 Amps. There are lots of USB-A chargers that deliver 3 Amps, so the setup will look like:

```
[Wall power to 3A USB] --USB to microUSB cable--> [TinyPilot power adapter] --USB-A to USB-C cable---> [Raspberry Pi USB-C]
```

ubenmackin, 2020-08-31:
> The previously attempted solution was to use a [USB-C OTG Y-cable](https://tinypilotkvm.com/assets/static/y-cable.2671d65.bf14ad9dfe719710c1f17d9fda9eb4fb.jpg) but has this problem (quoted from blog [...]

So currently I am doing this, but would like to understand more about what can happen and to which USB ports.

So I have the Y-Cable plugged into the Pi-4 USB-C, the monitored computer USB-A, and a [USB power device](https://www.amazon.com/Charger-RAVPower-Charging-Multiple-Compatible/dp/B00OQ19QYA/ref=sr_1_3?dchild=1&keywords=6+port+usb+power+supply&qid=1598904037&sr=8-3)

On the Y-Cable I bought, there are reviews of people using it in a [similar fashion](https://www.amazon.com/gp/customer-reviews/R1FK1N8UHXGACK/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B01L91CCHC). But I'm not an electrical engineer, so if this can cause issues, I'll stop using the cable and wait for the PCB solution.

mtlynch, 2020-08-31:
@ubenmackin - The issue is that USB ports are spec'ed to 5 V, but with tolerance of other voltages. The USB power spec allows voltages from 4.4 V to 5.25 V. If you plug the Y-cable into both a computer and an external power source, they won't have the exact same voltage. If the voltages are pretty close, it's not an issue, but if there's even a 0.1 V difference, it's enough to cause permanent damage to the computer's USB port or the Raspberry Pi. The problem is that you can't guarantee that the two different sources will maintain voltages that match closely enough to avoid damage.

I recommend powering exclusively from the computer power until a fix is ready. I'll have the PCB available for pre-order later this week. It's in production right now and could be ready to ship as soon as next week if testing is successful.

lrabe, 2020-09-01:
@mtlynch Will the PCB available on e.g. oshpark.com? What differences will there be to the USB/PWR splitter?

mtlynch, 2020-09-01:
@lrabe - To start out, I'm just going to make them available through the TinyPilot website, but I may evaluate other options later. I believe it's mostly similar to the USB/PWR splitter, except mine comes with a plastic case so that the PCB isn't exposed. Also, mine drops the barrel jack.

rvdwegen, 2020-09-02:
For anyone who's already ordered the Y cable from China I imagine you could also get there by carefully cutting the outside of the cable and snipping & taping off the power wires going to the PC side. Bit more r/techsupportgore but it should work right?

mtlynch, 2020-09-02:
>I imagine you could also get there by carefully cutting the outside of the cable and snipping & taping off the power wires going to the PC side. Bit more r/techsupportgore but it should work right?

@rvdwegen - Haha, I don't recommend that.

[image]

I've just put up the pre-order page for the power connector on the TinyPilot website:

<https://tinypilotkvm.com/product/tinypilot-power-connector>

mtlynch, 2020-09-09:
The official recommendation is now to use the TinyPilot power connector for continuous power. I've updated the README (#208), but let me know if you find other options that provide 3 Amps of power and mitigate the risk of reverse current.

closed 2020-09-09 by mtlynch