tiny-pilot/tinypilot #189: Waking up the target system with WOL

lrabe, 2020-09-01:
It would be nice to have a feature where the target system can be turned on from a shutdown state using Wake-on-LAN. 

Requirements: 
- Target system supports WOL (mostly motherboard and network card)
- Target system is on a wired connection, not on WLAN
- Target system and tinypilot device are in the same network/broadcast domain

I could probably implement this. There's a library to do the actual WOL stuff: https://pypi.org/project/wakeonlan/  

There are some design decisions to make though. For example, where is the MAC address stored?

- Option 1: MAC address has to be entered manually every time from the web interface. Might be inconvenient when the user doesn't know the MAC address but needs to start the system
- Option 2: MAC address is hardcoded in a config file. Simple solution, user only needs to configure it once
- Option 3: MAC address can be changed from web interface and is stored somewhere permanently. Elegant but slightly more complex solution. This would require some sort of permanent storage, which I don't think exists yet (?)

mtlynch, 2020-09-03:
Cool, this would be interesting. I'd like to go with the simplest solution first and then iterate on it, so Option 1 sounds best.

mtlynch, 2020-09-04:
Thinking about this more, I'd prefer to go the route of connecting directly into the target computer's motherboar. WOL requires fewer connections, but it's an incomplete solution since it can't turn off power or turn on motherboards that are disconnected from the network or don't support WOL.

closed 2020-09-04 by mtlynch

lrabe, 2020-09-08:
Connecting tinypilot to the motherboard makes this a way less portable/turnkey solution.  
I agree that the ability to turn the system off is an advantage generally, but imo it's not an ideal solution for all users. Normally, the system can be turned off from the OS.  
There might also be issues with warranty when opening up systems to connect tinypilot to the motherboard.

rvdwegen, 2020-09-08:
Crazy Idea but what about something like an ESP32 in the case connected over bluetooth?