tiny-pilot/tinypilot #193: Compute Moduel Support?

peterberbec, 2020-09-03:
The Pi CM3 is a Pi 3 in a SODIMM space. It seems, with the right size [carrier board](https://www.pishop.us/product/stereopi-slim-edition/), and some [software h4x](https://forum.stereopi.com/viewtopic.php?f=5&t=42&p=216&hilit=otg#p216), you can get OTG working with it. This, plus a [CSI to HDMI bridge](https://www.ebay.com/itm/For-Raspberry-Pi-HDMI-To-CSI-2-C779-Bridge-Adapter-Interface-Module-1080p-25fps/203009279885), would seem to allow a TinyPilot to be only slightly larger than a Pi Zero, making an even TinierPilot.

The Pi Zero [works](https://github.com/mtlynch/tinypilot/issues/93), but is very slow. I'm seeing close to 750ms of lag even when overclocking the CPU, GPU and RAM. It is my hope that using a CM3 will have a few advantages.

- Speed. 4 cores, much faster
- RAM. twice a Pi Zero
- Disk. A CM3+ can have much faster eMMC, so there is no SD card is involved
- Size, Size, Size. I dream of this being in a single, tiny, case.

There is a pretty big downside to non-[fullsized](https://www.pishop.us/product/stereopi-standard-edition/) version of this carrier board and it goes with the problem of the CM3

- No Network.

The Pi CM3 doesn't have wifi and the tiny carrier board doesn't have ethernet. Normally, I'd use g_ether, but since we are using g_hid to connect to the host machine, this won't work. It turns out, you can use GPIOs to connect to an [ESP32](https://hackaday.io/project/8678-rpi-wifi), but that's a lot of work. I'm just going to solder some [motherboard USB headers](https://www.amazon.com/dp/B000IV6S9S) onto the carrier board's unpopulated USB header and pop on some random [USB wifi dongle](https://www.amazon.com/Tenda-150Mbps-Wireless-Adapter-W311MI/dp/B006GCYAOI).

I have ordered a CM3+ 8GB eMMC Pi, the above carrier board, a wifi dongle and a motherboard USB header. I will report back.

mtlynch, 2020-09-03:
Cool! Yeah, I'm interested in supporting CM3. I'd expect it to work out of the box as long as you get Raspbian running and OTG enabled. If it doesn't work, I'm interested in what work is required to make it compatible.

peterberbec, 2020-09-03:
I'm obsessed with making a PicoPilot. 😀

Peter Berbec
(917) 213-2594

peterberbec, 2020-09-07:
> Cool! Yeah, I'm interested in supporting CM3. I'd expect it to work out of the box as long as you get Raspbian running and OTG enabled. If it doesn't work, I'm interested in what work is required to make it compatible.

I got the boards, but I guess I missed a part of the instructions, because I let out the magic smoke.

For anyone using this board, when you jump the two pins to enable flashing mode, make sure the switch is set properly or you'll burn out the carrier when 5v goes in some sort of loop.

WardBenjamin, 2021-02-04:
This could be tagged with alternate-equipment.