tiny-pilot/tinypilot #201: I'm having serious reliability issues

danmed, 2020-09-08:
Straight after install things seem to work great.. then i'll go back to it a week later and the page just endlessly loads.. Sometimes i get everything but the video output, sometimes it just stays completely white.

i reboot the pi, and the same thing happens again..

Normaly a couple of reboots or a new quickinstall will fix it for a while..

but this time i can't get any of it back..just got a white screen with a spinning wheel in the middle of the screen.

danmed, 2020-09-08:
As an aside, tonight i've done a apt update && apt upgrade.. and rebooted and i still can't get anything to work..

In chrome, i get everything looking nice, including the video, but it's not intercepting any of the commands (or atleast it's not updating the video).

In firefox the menu's are screwed up..

iexplore is just a no go completely.

Is there a way to uninstall completely and reinstall?

mtlynch, 2020-09-09:
Sorry it's gone so crazy. Can you share some logs?

```bash
sudo journalctl -u tinypilot
sudo journalctl -u nginx
```

Also, can you post screenshots of what you're seeing and any output you're getting in the developer console?

[image]

The cleanest way to start over would be to start with a fresh Raspberry Pi OS install and just run the quick-install.

danmed, 2020-09-09:
HI, so this morning in firefox it is taking forever to load, but when it does it looks OK, but i can't use the keyboard.

here are the logs : http://paste.debian.net/1163146/

[image]

Chrome loads much faster, but again I can't control anything.. I'll try and find some time later to redo the OS.

mtlynch, 2020-09-09:
It looks like TinyPilot is failing to send keystrokes and mouse events through the USB cable to the target computer. The delay you're seeing is probably TinyPilot getting backed up waiting on all the events to go through (it waits 0.5s for each keystroke or mouse movement to succeed). 

Can you try writing directly to the keyboard interface from the Pi on the command-line?

```bash
# Should type 'Hi' to the target computer.
echo -ne "\x20\0\xb\0\0\0\0\0" > /dev/hidg0 && \
  echo -ne "\0\0\xc\0\0\0\0\0" > /dev/hidg0 && \
  echo -ne "\0\0\0\0\0\0\0\0" > /dev/hidg0
```

I've seen the behavior you're experiencing when:

1. The USB cable between the Pi and the target computer is disconnected
1. The USB cable between the Pi and target computer doesn't support USB OTG (some cables don't)

It could also be that the cable got damaged (stepped on, rolled over by a chair). Do you have a spare cable you could test?

danmed, 2020-09-09:
So this is weird, i reseated everything and i it's working perfectly.. on 1 other machine i access it from.. from everything else it's still problematic..

All using firefox.. really not sure whats going on here.

mtlynch, 2020-09-09:
Can you share logs and screenshots from the machines where it's not working?

danmed, 2020-09-09:
The only thing that appears not to be working is keyboard input.. what logs do you want?

mtlynch, 2020-09-09:
Looks like it might be a caching issue. Can you hit Ctrl+Refresh in the browsers where it's not working and try again?

If that doesn't work, can you run this command from the Pi so that I can see what version you're running?

```bash
cd /opt/tinypilot
sudo git show | head -n 1
```

danmed, 2020-09-09:
While im waiting for it to load enough for me to do that, here's the output

commit 56af6cadca456b45881b69a3d30099decda87a6c

danmed, 2020-09-09:
Ah.. that might of worked... ctrl+f5.. thank you :)

danmed, 2020-09-09:
Hmm im still getting a lot of errors about the mouse though... is that normal?

mtlynch, 2020-09-09:
Can you try upgrading to the latest version with `quick-install`? I [disabled caching](https://github.com/mtlynch/tinypilot/issues/207) so we can rule that out as a possibility.

danmed, 2020-09-09:
run the upgrade and now nothing is being captured.. but it loads very fast lol

mtlynch, 2020-09-09:
Glad it's working!

closed 2020-09-09 by mtlynch