tiny-pilot/tinypilot #246: Mouse support is making Tinypilot unusable

JRansomed, 2020-09-18:
## Description

Anytime the mouse crossed the video part of the webpage keyboard it hangs everything up.  Can an option be added to disable mouse support.  Changing the MOUSE_PATH to something invalid doesn't stop the error from repeating a bunch of times and locking everything up.

## What's the behavior that you expect?

That the added mouse support wouldn't hang everything up.

## What's happening instead?

I can no longer send keyboard commands.  Whole thing seems to hang up.  The log starts rolling with:
`ERROR in main: Failed to forward mouse event: Failed to write to HID interface. Is USB cable connected?`
It will continue to repeat that a dozen or so times.  Until that stops, the Tinypilot is unusable.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Move the mouse into the video window
3. Watch the errors start rolling in journalctl -u tinypilot -f
4. Observe how nothing responds anymore

## Screenshots

## Logs

JRansomed, 2020-09-18:
If I comment out the mouse code in app.js, everything works as expected.  Something I just realized I should mention.  The Tinypilot is connected to a KVM.  Perhaps that is why the mouse code is not working.  I would rather be able to turn off the mouse code and use the KVM.  These servers are all text based, so 1 KVM to 1 Tinypilot is far more economical than a handful of Tinypilots with mouse support that I won't use.

Everything was working until I ran the script again to update.  I was looking forward to testing the paste functionality, but unfortunately got hung up with the mouse additions.

mtlynch, 2020-09-18:
Sorry about that! Thanks for reporting.

Oh, interesting about the KVM. What model? I wonder if they have trouble processing mouse events at the speed a computer does.

If you set `MOUSE_PATH` to `/dev/null`, the errors should stop. To permanently disable mouse support, you can do:

```bash
echo 'export TINYPILOT_INSTALL_VARS="tinypilot_mouse_interface=/dev/null"' >> ~/.bashrc
. ~/.bashrc
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

@djclueless is working on throttling down the number of mouse events that TinyPilot sends in #225, but we're still working on it.

Alternatively, it's possible that the bottleneck is the connection between your browser and TinyPilot. It seems unlikely given the errors you're seeing though. Are both on the same local network?

As a last resort, you can downgrade to a pre-mouse version of TinyPilot:

```bash
export TINYPILOT_INSTALL_VARS="tinypilot_repo_branch=ea853b54fa56ce275b6f776caa5a839451013e84"
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

JRansomed, 2020-09-18:
I tried setting the mouse interface to /dev/null and it did the same:
Tried /etc/environment  (Confirmed via env that the setting took)
Then tried hard setting it in main.py

It seemed the only way I was able to disable the error was to comment the mouse support out in app.js

The computer and TinyPilot are on the same network.  Currently the same VLAN, but that's not always the case when I access it.  While this is my home network, it's not a flat topology.  The gear is Cisco and Aruba, so pretty solid network connections here.

The KVM is a pretty cheap one from Amazon, $24 Nippon Labs KVM-4R-4FT:
https://www.amazon.com/dp/B01CJ0XZYQ

I've liked this model as it generally works pretty well.  It supports hotkey switching, which is surprising to find at it's price.  I've seen it get weird with PFsense (FreeBSD), but work just fine with my Debian and Ubuntu boxes.  I've been using this switch for about a year, but never to a system with a mouse.  Maybe it doesn't do well with mouse support.

Rather than downgrade, I commented out the mouse code in app.js.  I'd like to keep the clipboard functionality.  I'm happy to test if you have any other suggestions.   Thanks for all the suggestions you gave so far!

djclueless, 2020-09-19:
@mtlynch This happens to me a lot as well, and I am not using a KVM. For me its random, and sometimes it locks up the pi completely where I cant event ssh and it needs to be power cycled.

[image]

mtlynch, 2020-09-19:
Hmm, maybe we should switch to a ThreadPool instead of unconditionally spawning new threads for every keystroke or mouse event.

mtlynch, 2020-09-19:
I just merged in #248, which should prevent TinyPilot from locking up and might solve the issue entirely. I realized we don't need to use threads at all, so the writes are now less resource-intensive.

@JRansomed - It's surprising that you'd still see lockups when you set `MOUSE_PATH` to `/dev/null` because those writes should always succeed instantly. Are you running TinyPilot manually or through the systemd service? If you're running through the service, can you check your `/lib/systemd/system/tinypilot.service` and verify you see this line?

```
Environment=MOUSE_PATH=/dev/null
```

djclueless, 2020-09-19:
@mtlycn I have a new problem now not sure if its related to #248 , seems once I encounter one of those exceptions, now the keys repeat when pressed.  They key actually sends but I am getting a "failed to write key", and the key keeps repeating on the remote computer.

closed 2020-09-19 by mtlynch

JRansomed, 2020-09-20:
@mtlynch You hit it spot on.  I didn't think to look at what was configured in /lib/systemd/system/tinypilot.service.  It was not picking up my environment variable.

Tried some troubleshooting:  I checked env and it showed my setting.  Re-ran the install script, checked the systemd script, it did not update with /dev/null.  Tried deleting the systemd script and running the installer to see if it ansible would write a new file with my setting.  It did not.

For some reason it isn't picking up the environmental variable.  The installer is picking up that I'd like 720p, but for some reason MOUSE_PATH=/dev/null is being ignored.  I manually set the variable in the systemd script and it works!  Awesome!!!

This is a much easier work around.  Not sure why the install script is ignoring the environmental variable that all users get on my system.  My /etc/environment has:
```
export MOUSE_PATH=/dev/null
export TINYPILOT_INSTALL_VARS="ustreamer_resolution=1280x720"
```
Thanks again for all the hard work that has gone into developing this!

mtlynch, 2020-09-20:
@JRansomed - Oh, setting it in `/etc/environment` won't work if you're running it through systemd. Systemd processes have their own independent env variables, and you have to set them explicitly in the `tinypilot.service` file. If you change the `TINYPILOT_INSTALL_VARS`, it'll fix the environment variables in your `tinypilot.service` next time you run the `quick-install` script.

It should work if you change the line in `/etc/environment` to this:

```bash
export TINYPILOT_INSTALL_VARS="ustreamer_resolution=1280x720 tinypilot_mouse_interface=/dev/null"
```

Then, every time you upgrade, it'll preserve your settings. But I've added in throttling on the mouse events, so you might be able to run it with the default mouse settings and not see a slowdown if you upgrade to the latest bits.

JRansomed, 2020-09-20:
@mtlynch Welp, that would explain what I was doing wrong.  Thanks for the heads up!  : )

I ran the update script and removed any /dev/null settings.  The symptom has changed.  I don't see the mouse events anymore.  Unfortunately, if I cross the video screen with my mouse, TinyPilot stops working.  If I wait a little bit and refresh the page, I'll get a NGINX error.  Restart the TinyPilot service and it comes back.  As long as I avoid crossing the screen with my mouse, it'll work.  Thankfully, setting /dev/null the way you mentioned sets the variable and the problem goes away.

The log collection shows lots of connect failed errors in nginx.  An example is as follows:
```
2020/09/20 00:43:29 [error] 570#570: *187 connect() failed (111: Connection refused) while connecting to upstream, client: 10.126.38.33, server: tinypilot, request: "GET /socket.io/?EIO=3&transport=polling&t=NIem4Se HTTP/1.1", upstream: "http://127.0.0.1:8000/socket.io/?EIO=3&transport=polling&t=NIem4Se", host: "10.126.38.232", referrer: "http://10.126.38.232/"
```

If others aren't reporting this issue, it must be the cheap KVM causing my issues.  I don't want to be a bother and waste your time chasing something specific to me and my KVM.  Guess I've finally found something about it to point out to people if I recommend it.  At least setting the mouse path to /dev/null seems to pacify it.  : /

Again, thanks for all the hard work that you've poured into this.

mtlynch, 2020-09-20:
@JRansomed - oh, the lockup on mouse events is strange. I wonder if it's getting stuck attempting to write to the mouse interface. Would you be able to share your logs? If you run `/opt/tinypilot/dev-scripts/dump-logs`, it will prompt you to upload your recent TinyPilot logs and print a URL you can share.

JRansomed, 2020-09-20:
@mtlynch I ran dump-logs right after I hit the issue and stowed the log file away, just in case you wanted it.  I just uploaded it.  This should be pretty close to when I crossed the screen with the mouse.
[http://sprunge.us/A5HqCy](url)

Hopefully it has something interesting for you and not evidence of me making some foolish mistake.  ; )

mtlynch, 2020-09-21:
@JRansomed - I was able to reproduce the mouse lockup. I fixed it in #257, but I should find a way to improve it further. The problem is that TinyPilot can't tell whether its target machine accepts mouse input, so on mouse movement, it sends a bunch of mouse events and then gets backlogged waiting for them to complete, but each event takes 0.5s to time out, so it adds up. I'm not sure if the fix is to allow the user to disable the mouse or to just automatically stop trying to send mouse events after some number of failures.

JRansomed, 2020-09-22:
@mtlynch The fixes are definitely an improvement.  Unfortunately, the events generated by tracking the mouse movement across the screen can be quite a lot.  That makes the delay still noticeable.  It's not broken this way, that's a huge win for when someone doesn't know what's happening.  I, selfishly, like the idea of a way to turn the mouse off and on in the gui.  Admittedly, that is more a bandaid for the issue than a fix.  If another user hits this issue and doesn't realize its mouse movements, they are going to think TinyPilot doesn't work.

You mention automatically having it stop trying to send mouse events after some number of errors.  Is it cut and dry enough to see a threshold for that?  That might be more a self healing solution.  Would there be a way to then have it try again and see if mouse events work?  I'm wondering if there could be a scenario where the circumstances could change.  Maybe the TinyPilot is plugged into a KVM that works with one computer but not another?  This does not appear to be the case with mine, so this could be completely off the mark.  There's appeal to it attempting to back off when it determines a problem (detects too many errors) and restoring functionality if that condition goes away.

Could have the option to enable/disable the mouse in the gui with an indicator as to its present state.  Have a threshold where too many events disables the mouse and have it set the mouse to off and set the indicator in the gui to inform the user.  Maybe a message to the user to inform them that the mouse was disabled due to errors, but that they can attempt to re-enable it?

For my present setup, it will still make sense to disable the mouse events as the delay is noticeable.  I just have to skim a part of the screen, which with multiple monitors is easy to do when you want to cross to another.

Hope my rambling is more helpful than a waste of time.  This turned out awfully wordy.  Thanks again for this awesome project!

djclueless, 2020-09-22:
Just curious how do you have your tinypilot connected to the network? Wired or wireless?

Can you confirm that the socket server is actually hanging up and not the browser struggling to pull in the ustreamer images?
If I force network speed down to 3G on chrome, the mouse and keyboard events work, but video stream laggs or just stops all together.

Anyway to monitor your keystrokes on the remote computer once the tinypilot has locked up to see if they actually go through?
A second monitor or maybe at the console just type commands to see if they are there next time you reboot the tinypilot?

mtlynch, 2020-09-22:
@JRansomed - thanks, those thoughts are helpful!

I think there are two main problems:

1. When TinyPilot fails to forward mouse events, it doesn't give any visible notice to the user.
1. When TinyPilot fails to forward mouse events, it keeps trying to send them at the same rate.

I'd like to explore adaptive throttling for mouse events. I think that will improve performance in general. Something like exponential backoff in TCP. If we're getting fast responses for mouse events, we should send faster to take advantage of the target computer's responsiveness. If they're clogging the socket or failing, we should keep backing off.

I'll file separate issues to track these work items.

mtlynch, 2020-09-23:
Okay, tracking these issues in two new bugs:

* Indicate in web UI when forwarding a mouse event failed #267
* Adaptive throttling for mouse events #268

PR #250 (2020-09-19): Throttle mouse events
Fixes #246