tiny-pilot/tinypilot #270: Add support for mobile devices as clients

ksherwood02, 2020-09-23:
When using a mobile device (Chrome on Android, Chrome or Safari on iOS13) only the initial input is registered.  I'd love to contribute and I'll start digging in to this if I can, but I'm sure others would benefit from this too.

mtlynch, 2020-09-23:
Thanks for filing this, @ksherwood02! Can you share more details? I'm not sure what you mean when you say "only the initial input is registered."

Support for input from mobile devices was only added this morning (#266). Are you using the latest version?

ksherwood02, 2020-09-23:
If I bring up MS Paint, as an example app so I can see where the cursor is, anywhere I touch the cursor moves, so it's registered the touch, but not the "click."  Beyond that, I can't drag my finger across the screen to indicate the mouse has moved either.  As I think about it, converting from a touch interface - deciding what is a click, a move, a click/drag, etc., will probably be a bit more involved.

When I use Chrome on Android, it registers a point being dropped where I touch.  When I use Chrome on an iPad, it moves the curosr w/o placing a dot.
The dots are from controlling it through Android.

[cid:88838f39-d603-44ba-9cf0-3f65ee9ae3f5]

Thanks,
Kevin

mtlynch, 2020-09-24:
Ah, gotcha. Someone started working on this in #192 but then work stopped. It doesn't look like it'd be too much work to get touch events working.

db39, 2023-01-24:
[A user on the forum](https://forum.tinypilotkvm.com/-525/tiny-pilot-access-from-mobile-device) has a use case that would take advantage of touch screen input (accessing the TinyPilot web interface remotely using Tailscale on a smartphone).

db39, 2023-02-09:
Some potentially useful info: Firefox's responsive mode has a 'touch simulation' option which could be very useful for testing touch functionality on desktop.

mtlynch, 2023-09-21:
Adding references to some draft PRs around this that never got merged:

https://github.com/tiny-pilot/tinypilot/pull/1387
https://github.com/tiny-pilot/tinypilot/pull/1385

db39, 2024-01-15:
Another [user on the forum](https://forum.tinypilotkvm.com/-865/controlling-using-android-phone-or-ipad) is also interested in touch screen support.