tiny-pilot/tinypilot #273: German Keyboard Layout?

pilo1337, 2020-09-29:
Hi Guys!

Awesome oroject, works flawlessly, thanks!

But one Question remains:

is it possible to add a german keyboard layout?
Exp. the function of "Alt Gr" could be a bit complicated?

Thanks!

mtlynch, 2020-09-29:
I'll eventually add support for this, but I don't have a timeline. I'd have to purchase a QWERTZ keyboard first.

If you're interested in contributing, it shouldn't be too difficult. You'd have to:

1. Add a qwertz.py file to [/app/hid/keycodes](https://github.com/mtlynch/tinypilot/tree/d7439638e5279164db04c474ede38c4f9ce21996/app/hid/keycodes)
1. Update [/app/js_to_hid.py](https://github.com/mtlynch/tinypilot/blob/d7439638e5279164db04c474ede38c4f9ce21996/app/js_to_hid.py) to add a reference to qwertz and any new mappings.
1. Update [/app/static/js/keycodes.js](https://github.com/mtlynch/tinypilot/blob/d7439638e5279164db04c474ede38c4f9ce21996/app/static/js/keycodes.js#L134) to include `de` as a browser language where we assume an Alt Gr key exists.

mtlynch, 2020-10-09:
I added a few more details to this process here:

https://github.com/mtlynch/tinypilot/wiki/Add-support-for-a-new-keyboard-layout

mtlynch, 2020-12-09:
This should now be fixed with #355

closed 2020-12-09 by mtlynch