tiny-pilot/tinypilot #276: Support audio

mtlynch, 2020-10-01:
I'm not sure how feasible this is given TinyPilot's current recommended hardware. If we can't forward the sound, it might be sufficient to detect that a sound event occurred on the remote machine and somehow give a visual indicator in the TinyPilot interface.

cagnulein, 2021-07-12:
so no sound at all? doesn't the input hdmi usb support this?

mtlynch, 2021-07-12:
Some HDMI capture devices support sound, but we also need to pipe sound through the rest of the system. We're transmitting the remote screen in MJPEG format, which doesn't support sound.

mtlynch, 2022-03-21:
We're in the process of working on H264 support in #875

1nnovat1on, 2022-12-30:
I also would greatly benefit from this feature

mtlynch, 2023-03-02:
This is now fixed and available in Voyager 2a devices or DIY devices that have audio-aware HDMI capture.

closed 2023-03-02 by mtlynch