tiny-pilot/tinypilot #286: Data USB / keyboard not working

alex-phillips, 2020-10-09:
## Description

When I plug in the 'data' USB into the computer, the terminal outputs the following and no input is detected:

[image]

## What's the behavior that you expect?

USB to allow keyboard input

## What's happening instead?

See screenshot above.

## What are the steps to reproduce this behavior?

1. Plug data USB into machine (linux, ubuntu 20.04 specifically)
4. See error

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`

http://sprunge.us/ILKTaM

**NOTE**: I am using a spliced USB cable following the diagram below. This is due to the splitter being backordered and other options taking a month or longer to ship.

[image]

djclueless, 2020-10-09:
When you spliced the cables, did you use common ground

The usb host and the power supply ground cables need be spliced together.

alex-phillips, 2020-10-09:
Yesh - from the Pi is red (power), green (data +), white (data -), and black (ground). I spliced from the power supply, red and black to the USB cable and to the host, I spliced white, green, and black.

I will say that the USB C cable I used didn't have a 'black' ground but it appears its ground was unshielded and loose so assumed this was the ground whereas the cable from the power supply specifically has a black wire (this was not a USB-C cable but a USB-A cable I just needed for the PSU end).

alex-phillips, 2020-10-09:
@djclueless so all three connections are sharing a twisted black or unshielded wire. All other wires are only shared by 2 of the 3 ends.

djclueless, 2020-10-09:
Do you have an unmodified USB-C cable to test?

Just connect the PI to the computer using a regular cable.  See if the keyboard works? This would be the easiest way to figure out if its a cable issue.

I spliced my own cable as well, so yours should look similiar to the picture below (I know it's messy but it works). Basically what you explained.

From Power supply you have:
- Positive (red to USB-C)
- Negative (black or shield or white in my case to both the USB-A and USB-C).

From USB-C (PI) you have:
- Positive (red from p/s)
- Negative (black or shield from p/s & negative to USB-A host)
- Data cable + (White or Green to usb-a host)
- Data cable - (White or Green to usb-a host)

From USB-A (Host) you have:
- Negative (Black or shield to USB-C and p/s)
- Data cable + (White or Green to USB-A)
- Data cable - (White or Green to USB-A)
- No Positive (Red cable should be cut and not connected to anything)

[image]

alex-phillips, 2020-10-09:
@djclueless Thanks for the info! I've attached screenshots of my three cable splices. Mine's a lot messier, but wanted to get it working before I cleaned it up. One thing I noticed is that the USB-A cable that is connected to the  power supply has both a black and an unshielded wire, which I currently don't have connected to anything. Any idea what that would be if it also has a specific black wire?

PSU Wire:
[image]

Pi Wire:
[image]

Host wire:
[image]

mtlynch, 2020-10-09:
>This is due to the splitter being backordered and other options taking a month or longer to ship.

Just to clarify, everything is currently in stock. You saw things listed as out of stock on [this page](https://tinypilotkvm.com/products)?

Did the USB-C cable work before you spliced it? There are some that just don't work with USB OTG for whatever reason, including Amazon Basics brand ones.

alex-phillips, 2020-10-09:
@mtlynch When I checked in September it said it was backordered. Glad to see everything's back in stock!

@djclueless  But it does look like it's a cable issue. I tried with an unmodified USB-C cable (same brand and type) and it doesn't seem to be working. I'll test it out with another and report back.

Thank you both for helping troubleshoot!

alex-phillips, 2020-10-09:
@mtlynch is there a specific brand you'd recommend getting? It seems none of the USB-C cables I have are working.

mtlynch, 2020-10-09:
@alex-phillips - For a regular USB-C to USB-A connection between the Pi and the target computer, I know [Anker](https://smile.amazon.com/gp/product/B07DD5YHMH/) works reliably. I've never tried any DIY splicing with it though.

If you want to skip the DIY splicing, I recommend:

* [TinyPilot power connector](https://tinypilotkvm.com/product/tinypilot-power-connector)
* [CableCreation 6" USB-C to USB-A](https://smile.amazon.com/gp/product/B012V56D2A/)
* [Rankie 3' MicroUSB to USB-A](https://smile.amazon.com/gp/product/B01JPDTZXK/) (it's a 3-pack, because you need two of these)

alex-phillips, 2020-10-09:
@mtlynch Interesting. Both the cables I've tried are Anker (skipping the splicing).

alex-phillips, 2020-10-09:
Let me try another server just in case there's something unusual about this motherboard...

conrad784, 2020-10-14:
I found your issue, because I had the same error message. For my Supermicro board it worked by using the USB2.0 ports instead of 3.0.

mtlynch, 2020-10-30:
@alex-phillips - Are you still seeing this behavior or did new cables resolve this?

mtlynch, 2020-11-04:
Closing for now, but feel free to reopen if this is still an active issue.

closed 2020-11-04 by mtlynch