tiny-pilot/tinypilot #287: Install Script / Ansible Playbook can't finish - Fails

SmugZombie, 2020-10-09:
## Description

When attempting to run the install script from the main readme (curl quick-install). The script runs to a certain point then fails.

[image]

Returns an error that python-pip is not available. Which is true.. Because it is python3-pip

## What's the behavior that you expect?

Installation to look for the proper package and install properly

## What's happening instead?

Installation is looking for an invalid package

## What are the steps to reproduce this behavior?

Ran the quick-install script multiple times..

[image]

## Screenshots

Seen above

## Logs

<pre>root@arm-cluster-01:/# curl   --silent   --show-error   https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install |     bash - &&   sudo reboot
+ set -e
+ '[' -f /etc/nginx/sites-enabled/KVMPi.conf ']'
+ '[' -z '' ']'
+ TINYPILOT_INSTALL_VARS=
+ echo 'Using default install vars'
Using default install vars
+ set -u
++ mktemp -d
+ pushd /tmp/tmp.PVrdJM7deo
/tmp/tmp.PVrdJM7deo /
+ sudo apt-get update
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
+ sudo apt-get install -y libffi-dev libssl-dev python3-dev python3-venv
Reading package lists... Done
Building dependency tree
Reading state information... Done
libffi-dev is already the newest version (3.3-4).
libssl-dev is already the newest version (1.1.1f-1ubuntu2).
python3-dev is already the newest version (3.8.2-0ubuntu2).
python3-venv is already the newest version (3.8.2-0ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 25 not upgraded.
+ python3 -m venv venv
+ . venv/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/tmp/tmp.PVrdJM7deo/venv
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++ PATH=/tmp/tmp.PVrdJM7deo/venv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++ export PATH
++ '[' -n '' ']'
[... 169 lines cut ...]
ok: [localhost]

TASK [mtlynch.tinypilot : enable dwc2 driver in boot config] *******************
skipping: [localhost]

TASK [mtlynch.tinypilot : enable dwc2 driver in modules] ***********************
ok: [localhost]

TASK [mtlynch.tinypilot : install usb-gadget initializer as a service] *********
ok: [localhost]

TASK [mtlynch.tinypilot : enable systemd usb-gadget initializer service file] ***
ok: [localhost]

TASK [mtlynch.tinypilot : install TinyPilot pre-requisite packages] ************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'python-pip' is available"}

PLAY RECAP *********************************************************************
localhost                  : ok=30   changed=0    unreachable=0    failed=1    skipped=12   rescued=0    ignored=0
</pre>

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`

`root@arm-cluster-01:/# cat /opt/tinypilot/dev-scripts/dump-logs
cat: /opt/tinypilot/dev-scripts/dump-logs: No such file or directory`

SmugZombie, 2020-10-09:
While I now see my issue is likely that I am running Ubuntu and not Raspbian.. I am guessing this is a relatively easy fix (if only I knew ansible)

mtlynch, 2020-10-10:
I messed around with it a little bit and got it past the python-pip failure:

https://github.com/mtlynch/ansible-role-tinypilot/pull/56

The problem is that I don't know how to get the ConfigFS HID gadget to load on Ubuntu. So everything installs, but TinyPilot still can't send keyboard or mouse input to the target computer. I tried adding `dtoverlay=dwc2` to `/boot/firmware/usercfg.txt`, but that sent it into a boot loop (on Raspberry Pi OS the role [automatically adds this line](https://github.com/mtlynch/ansible-role-tinypilot/blob/7f7204a7137ec5d0a5082d1f4c796e581df3deb1/tasks/install_usb_gadget.yml#L7L12) to `/boot/config.txt`).

I've tried on [a few different SBC OSes](https://github.com/mtlynch/tinypilot/issues/79#issuecomment-663736165), and Raspbian is the only one where I've been successful.

mtlynch, 2021-01-16:
This is now working as of this change:

https://github.com/mtlynch/ansible-role-tinypilot/pull/77

closed 2021-01-16 by mtlynch