tiny-pilot/tinypilot #302: Replace manual modifier buttons with on-screen keyboard

mtlynch, 2020-10-19:
To allow the user to send key combinations that the OS or browser would otherwise trap (e.g., Ctrl+Alt+Del, Ctrl+T), we use manual modifier buttons below the remote screen:

[image]

These buttons aren't especially intuitive. It would be better if we replaced them with a full keyboard, implemented in HTML/CSS/JS. It would need standard QWERTY keys plus left shift, right shift, left alt, right alt, etc. for users who have applications that are sensitive to left vs. right modifier keys.

manuschillerdev, 2020-11-12:
I was playing around with something very similar recently.
Just leaving this here in case it may help someone as starting point to implement your feature.

https://codesandbox.io/s/modest-breeze-3vl59?file=/src/index.js

matt-curtis, 2020-11-16:
Was going to give this a shot, looks like someone's already working on it. If it opens back up let me know!

PR #333 (2020-11-17): On screen keyboard
Explanation of the changes, what they achieve, and any design decisions that are worth mentioning.

Fixes #302 

Replace keyboard panel with a full html & css keyboard.

**Delete this line and everything after before submitting your PR**

---

## How to get your PR merged quickly

* Give a clear, one-line title to your PR.
  * Good: `Fix dropped keystrokes on Firefox`
  * Bad: `Fix issue`
* Title your PR in the present tense.
  * Good: `Change the background color to hot pink`
  * Bad: `Changing the background color to hot pink`
  * Bad: `Changed the background color to hot  [... cut]
closed 2020-11-17 by mtlynch