tiny-pilot/tinypilot #309: Keys not registering, or registering incorrectly

blalor, 2020-10-27:
I'm using macOS 10.14.6.

On Safari 14.0: multiple keys don't register presses; typing `fusion engine` in tinypilot becomes `uion ngin` on the console.

Firefox 81.0.2: `-` becomes `=`, `_` becomes `+`; typing `foo-bar_baz` becomes `foo=bar+baz`.

This is after loading the tinypilot page, or reloading it.  No messages appear in the browser console.

## Logs

http://sprunge.us/DyzUXL

mtlynch, 2020-10-28:
Thanks for filing this, @blalor!

What kind of hardware / OS is the TinyPilot plugged into? When you say macOS 10.14.16, that's the machine from which you access the TinyPilot web UI, correct?

There are two interesting things in your logs. The first are these lines:

```text
Oct 27 15:26:56 tinypilot python[2820]: [2020-10-27 15:26:56,800] WARNING in socket_api: Unrecognized key: Meta (keycode=224)
```

I've filed #310 to track that.

The second are these lines, which are likely related to the problem you're seeing:

```text

Oct 27 15:27:51 tinypilot python[2820]: [2020-10-27 15:27:51,529] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Oct 27 15:27:53 tinypilot python[2820]: [2020-10-27 15:27:52,054] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Oct 27 15:27:53 tinypilot python[2820]: [2020-10-27 15:27:52,578] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Oct 27 15:27:53 tinypilot python[2820]: [2020-10-27 15:27:53,113] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
```

`/dev/hidg1` is TinyPilot's mouse interface. These errors suggest that the target computer sees TinyPilot as a keyboard even though it's trying to present as a multi-function keyboard and mouse device. I've seen this when I connect TinyPilot to another KVM, but I haven't seen this when connecting directly to another machine.

blalor, 2020-10-29:
Yes, I'm accessing the TinyPilot UI from a Mac.  The target host is a Dell running Ubuntu 18.04, running in multi-user mode (rather than graphical).  

I haven't needed ⌘ (Meta), yet, but it's possible I hit that trying to reload the page, or just a fat-finger.  Thanks for catching that, tho.

When I reported this, I still had a physical keyboard and mouse attached to the target computer, as well as a Yubikey, in addition to TinyPilot, but with those disconnected I still have the missing and mis-mapped characters.  Those `Failed to write to HID interface` errors do appear consistent with trying to type `e` and `s` via the TinyPilot UI.

mtlynch, 2020-10-29:
>The target host is a Dell running Ubuntu 18.04, running in multi-user mode (rather than graphical).

Oh, that explains the `/dev/hidg1` errors. The Ubuntu 18.04 doesn't accept mouse input, so that's why those log messages appear. They're likely just appearing whenever your mouse cursor crosses the screen because TinyPilot is attempting to forward them to the target.

>I still have the missing and mis-mapped characters. Those Failed to write to HID interface errors do appear consistent with trying to type e and s via the TinyPilot UI.

Oh, I didn't realize it was always the same characters consistently. Is the Ubuntu system configured for a non-QWERTY keyboard? Can you share the output of `cat /etc/default/keyboard`?

If you visit [keycode.info](https://keycode.info/) from your Mac, do the keys come up correctly? You should see:

| Key | Code |
|------|------|
| `e` | `69` |
| `f` | `70` |
| `s` | `83` |
| `-` | `189` |
| `+` | `189` |

Lastly, could you try [enabling debug logging](https://github.com/mtlynch/tinypilot/wiki/Troubleshooting-and-Diagnostics#enable-debug-logging) temporarily? If you could reproduce the issues with the mistyped keys and then re-run `/opt/tinypilot/dev-scripts/dump-logs`, I can pinpoint where the keys are getting mistranslated.

Sorry for the hassle, and thanks for helping me track this down!

blalor, 2020-10-30:
So this is interesting. keycode.info is _also_ not registering several keys: e, s, d, f.  They *do* register when combined with (left) shift. 

For the other two:

| key     | safari event.key | event.which | event.code | firefox event.key | event.which | event.code | 
| ------- | ---------------- | ----------- | ---------- | ----------------- | ----------- | ---------- |
| `-`       | `-`                | `189`         | `Minus`      | `-`                 | `173`         | `Minus`      |
| `shift+-` | `_`                | `189`         | `Minus`      | `_`                 | `173`         | `Minus`      |
| `=`       | `=`                | `187`         | `Equal`      | `=`                 | `61`          | `Equal`      |
| `shift+=` | `+`                | `187`         | `Equal`      | `+`                 | `61`          | `Equal`      |

I observed the same behavior for all keys using an external keyboard, as well, and it registers `-`, `+`, and `=` as separate `Numpad…` codes.

Having another issue at the moment, I'll get the debug logs soon.

blalor, 2020-10-30:
logs: http://sprunge.us/JW7qvy

mtlynch, 2020-10-30:
Ah, okay. It looks like there was a bug in Firefox's key mapping. Fixed in #311.

If you run the auto-installer, it will pull down the latest version:

```bash
/opt/tinypilot/quick-install
```

The behavior on Safari, I can't figure out what's going on. If it fails on keycode.info, it's not going to work with TinyPilot because the browser just isn't generating events for those keys. Do you have another Mac you can test on to see if it's a general Safari problem or potentially an issue with that particular Safari instance's configuration?

mtlynch, 2020-11-04:
Hi @blalor - just wanted to follow up. Were you able to achieve expected behavior with the new update? Do you see the issue if you try Safari from another machine?

blalor, 2020-11-04:
I’m sorry, I haven’t had a chance. I’ll try to test it out this evening.

blalor, 2020-11-10:
The keys are mapping correctly on Firefox, now. 

As for Safari, I've tried keycode.info on a couple of different computers with the same result.  http://www.javascriptkeycode.com does work, however.  I'm not able to figure out what the implementation difference is, however (your code is much easier to follow!).

mtlynch, 2020-11-10:
Oh, that's strange. I don't have an OS X machine available, so it's hard for me to test. I just tried on Browserstack, but I got kind of bizarre results, and I'm thinking Browserstack is probably doing its own keyboard emulation.

Are those four keys the only ones it gets wrong? Because we could just put in a workaround that normalizes those keystrokes when the user accesses TinyPilot via Safari.