tiny-pilot/tinypilot #313: Take screenshots of remote machine

mtlynch, 2020-11-04:
Some users have requested the ability to save screenshots of the remote machine. This could be a neat feature.

Current workaround: Take a screenshot of the browser window and crop it to just the remote screen portion.

djclueless, 2020-11-08:
I have thought about this too, but I have just been using snip tool on my remote machine.

This would be the easier / fastest solution. I can have this done today if you are good with it.

- You need to add /snapshot in nginx 
**/etc/nginx/sites-enabled/tinypilot.conf**
```
    location /snapshot{
      proxy_pass http://ustreamer;
    }
```

**app\templates\components**
```
<li>
          <a id="snapshot-btn">Snapshot</a>
</li>
```

**app\static\jsapp.js**
```
document.getElementById("snapshot-btn").addEventListener("click", (evt) => {
  snapShot();
  evt.preventDefault();
});
```

**app\static\jsapp.js**
```
function snapShot() {
	var link = document.createElement('a');
	link.href = '/snapshot';
	link.download = 'TinyPilot-'+new Date().getTime()+'.jpg';
	document.body.appendChild(link);
	link.click();
	document.body.removeChild(link);
}
```

mtlynch, 2020-11-09:
Oh, I like that solution! I just [updated the nginx config](https://github.com/mtlynch/ansible-role-tinypilot/pull/58) to forward calls. I agree with the rest of your proposal, too. My only note is that I'd prefer to use the term "screenshot" as opposed to "snapshot."

PR #319 (2020-11-10): Add screenshot option
Add screenshot option

Fixes #313
closed 2020-11-10 by mtlynch