tiny-pilot/tinypilot #314: Is Xorg process used by TinyPilot?

MikeRotch76, 2020-11-05:
I have a program that graphs among other things, CPU usage.  I upgraded to TinyPilot v 1.1.3 and my history has shown that since then (I think), my avg CPU load is sitting around 1, whereas before it was hovering near 0.  Now, I can't remember exactly when I updated from 1.1.0 to 1.1.3.  But I think it was around Oct 31/Nov 1 and that's when my history shows the jump to 1 load avg.  Now, I have also recently updated the software on my rpi, but the thing is, I can't remember when I last ran apt upgrade, heh.  So I am wondering if the jump in CPU usage of the Xorg process is due to TinyPilot, or something else.  

The only other software I have running is AdGuardHome, and a cron script that uses curl to see my WAN IP for Dynamic DNS. I am not a linux or rpi expert but to the best of my knowledge, Xorg a process that has something to do with the desktop GUI. 
 Neither of those programs would use xorg.  I don't have a display plugged into the rpi.  The only thing I have plugged into USB is the tinypliot power device which was installed way before Nov 1st.

Before I cut a ticket on this, could TinyPilot be the cause of my Xorg spike?
[image]

MikeRotch76, 2020-11-05:
[image]

mtlynch, 2020-11-05:
It shouldn't be. TinyPilot doesn't use Xorg directly at all. It's possible that some TinyPilot dependency pulled it in indirectly, but it's unlikely, since TinyPilot runs in console-only mode without an issue.

closed 2020-11-05 by mtlynch