tiny-pilot/tinypilot #315: Connection on wifi is almost unusable while on ethernet is butter smooth.

ramj9, 2020-11-06:
Connection on wifi is unusable while on ethernet it is butter smooth. 

Could this be taken up as an issue?

mtlynch, 2020-11-06:
Thanks for reporting this!

What bandwidth do you get on WiFi on the device outside of TinyPilot? Can you run a speed test between your Pi and the computer you're trying to access it from?

ramj9, 2020-11-07:
```
pilot@tinypilot:~ $ iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
^T[  4] local 192.168.20.99 port 5001 connected with 192.168.20.25 port 62392
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-14.8 sec  19.5 MBytes  11.1 Mbits/sec
[  4] local 192.168.20.100 port 5001 connected with 192.168.20.25 port 62400
[  4]  0.0-10.7 sec  12.8 MBytes  9.99 Mbits/sec
```
Not considerably different. Top one is ethernet @11.1 Mbits/sec and bottom one is on wifi (192.168.20.100) @9.99 Mbits/sec

mtlynch, 2020-11-10:
That's strange. I'm not sure why there'd be a difference in WiFi and LAN if they get roughly equivalent bandwidths.

Is there something strange about your network setup? 10 Mbps over WiFi is pretty slow, and 10 Mbps over wired ethernet is **extremely** slow. Do you know why you're seeing speeds like that?

closed 2020-11-12 by ramj9