tiny-pilot/tinypilot #316: Ubuntu Server 20.04.01 LTS - Simple Install Fails

jharmon96, 2020-11-06:
## Description

Simple Install Fails on Raspberry Pi 4 - 8GB running Ubuntu Server 20.04.01 LTS

**Error message:**
TASK [mtlynch.tinypilot : install TinyPilot pre-requisite packages] ************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'python-pip' is available"}

## What's the behavior that you expect?

Install to complete

## What's happening instead?

Install fails. 
Port 80 returns Nginx 502 Bad Gateway

## What are the steps to reproduce this behavior?

Build SD image with Raspberry Pi Installer

## Screenshots

*If applicable, add screenshots to help explain your problem.*
[image]

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`

N/A

jharmon96, 2020-11-06:
Just noticed Ubuntu isn't supported per the requirements. You can change this to a Feature Request.

mtlynch, 2020-11-06:
Thanks for reporting this! Yeah, Ubuntu is unfortunately not supported at the moment.

#287 tracks the feature request to support Ubuntu.

closed 2020-11-06 by mtlynch

mtlynch, 2021-01-16:
Update: this is working now, as of this change:

https://github.com/mtlynch/ansible-role-tinypilot/pull/77