tiny-pilot/tinypilot #317: Feature Request: Low bandwidth mode

nolanl, 2020-11-07:
I just painfully fixed a server issue over the internet from a crappy hotel wifi connection. The video would fall behind enough that I had to type in a command, reload the page to see that I had typed it correctly, then hit enter.

It would be super cool to have a "low bandwidth" checkbox (or better yet, detect it and switch automatically, with some visual indication that the switch has happened). The idea being to switch to transcoding to whatever the best codec a pi4 can handle. The latency hit of transcoding is trivial compared to the stream continuously falling further and further behind.

mtlynch, 2020-11-10:
Yeah, this is something on my radar. There are a few things we can do to improve this.

One is making video stream settings dynamic instead of static. Currently, the device gets a static configuration for capturing video, and there's no way to change things like resolution or frames per second without reinstalling with different options. There's nothing technically preventing us from changing settings on the fly in response to changes in the UI or even auto-optimizing, but we just need to put in the plumbing to make that happen.

The other is changing video codecs. TinyPilot currently uses uStreamer, which outputs only MJPEG. MJPEG is pretty bandwidth-hungry because it's just sending a series of JPEG images. Switching to a format like x264 would likely trim down bandwidth, but it requires either switching streaming backends or adding a new codec option to uStreamer.

mtlynch, 2021-05-18:
This is fixed as a side effect of #448. Switching to h264 is still on the roadmap, which will hopefully make low-bandwidth scenarios even better, but for now, the video streaming options allow a lot of control over bandwidth consumption.

closed 2021-05-18 by mtlynch