tiny-pilot/tinypilot #344: Blank screen with no useful information

tianhuxu, 2020-11-25:
I followed the instructions and set up my raspberry pi 4, however, it always show a screen like this:
[image]

My MB is Asus z97i-plus, and the host OS is unRaid. What could cause this issue?

mtlynch, 2020-11-25:
That's strange. I've never seen that before. If you reboot the unRAID server while the Pi is connected, does anything change?

Can you share the logs from this command?

```bash
/opt/tinypilot/dev-scripts/dump-logs
```

cradcore, 2020-12-11:
I am also getting the same screen. Here is the bug report: http://sprunge.us/7NjtUK

I'm guessing this is the problem, although all my cables are plugged in:

`ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?`

Any ideas?

cradcore, 2020-12-11:
Update: I moved the dongle to a different USB (3.0 to 2.0) port, and the above error no longer appears. Unfortunately, the rainbow screen, still does. See http://sprunge.us/xTKLC7

mtlynch, 2020-12-11:
@cradcore - Thanks for reporting! What's the hardware and OS of the system TinyPilot is connecting to?

cradcore, 2020-12-12:
The machine is running Ubuntu 18.04. Below are the hardware specs

[image]

mtlynch, 2020-12-18:
Sorry, I can't figure what's wrong. It might just be that the dongle isn't compatible with this particular hardware.

What's the graphics card? Does keyboard forwarding work? Or do you just not have any display hooked up?

cradcore, 2020-12-20:
There is no dedicated graphics card in this machine, its a headless server. The main reason I wanted the KVM is because often times I have to change boot parameters on the machine. I also have no display hooked up at the moment, usually only when I need to change something in the BIOS

mtlynch, 2020-12-21:
Can you verify that you're getting output when you hook up a physical monitor? I suspect that the rainbow screen is just what's coming out of the video output port because I don't think that TinyPilot will generate that on its own.

PaperLawyer, 2021-01-07:
I sometimes have the same rainbow screen when booting my DIY NAS with an N3700-ITX motherboard until linux itself is booting i.e. on certain occasions (but not all) I cannot see the BIOS.

cradcore, 2021-02-05:
@mtlynch Sorry its been so long, I haven't had a chance to look at this again. When I connect the machine to a physical monitor, it shows the bios when booting and the terminal after it boots.

mtlynch, 2021-02-08:
Hmm, I think it might be a problem in the EDID, the monitor functionality that TinyPilot advertises to the target computer. Could you try applying the EDID below and let me know if it works?

```bash
export TINYPILOT_INSTALL_VARS="ustreamer_edid=00ffffffffffff0052628888008888881c150103800000780aee91a3544c99260f505425400001000100010001000100010001010101d32c80a070381a403020350040442100001e7e1d00a0500019403020370080001000001e000000fc0050492d4b564d20566964656f0a000000fd00323d0f2e0f000000000000000001c402030400de0d20a03058122030203400f0b400000018e01500a04000163030203400000000000018b41400a050d011203020350080d810000018ab22a0a050841a3030203600b00e1100001800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000045"
/opt/tinypilot/scripts/upgrade && sudo reboot
```

closed 2021-03-09 by mtlynch