tiny-pilot/tinypilot #352: HID mode not supported on IHSE KVM

bkortleven, 2020-12-03:
## Description

When connecting the TinyPilot to an IHSE Draco KVM CON device (console) which is connected to our IHSE KVM matrix, keyboard and mouse information is not (correctly) passed through the CON unit to the computer on the other side.
A test Windows 10 host reports 'Unknown USB device (invalid configuration descriptor)'.

## What's the behavior that you expect?

Windows 10 should see the normal HID profile, just like a direct connection of a keyboard/mouse on the CON unit would provide.

## What's happening instead?

A test Windows 10 host reports 'Unknown USB device (invalid configuration descriptor)

## What are the steps to reproduce this behavior?

Selecting a host extension on the KVM (via another CON/management console as the keyboard Shift-Shift-O isn't passed correctly)

## Screenshots

N/A

## Logs

 http://sprunge.us/9SuIS8

mtlynch, 2020-12-04:
Thanks for reporting this! Sorry about the issue.

There's a bug in the [HID descriptor](https://github.com/mtlynch/tinypilot/blob/2d086e5d8f970b813afd3106f949f247add2d31b/scripts/usb-gadget/init-usb-gadget), and I haven't been able to figure out what the issue is. I'll spend some time debugging next week to see if I can get to the bottom of this.

mtlynch, 2020-12-10:
Hmm, after digging into it more, what I thought was a bug in the HID descriptor may actually be a bug in the Linux USB gadget driver that makes it incompatible with certain USB hosts.

I tested TinyPilot's HID descriptor in a few different tools, and they all show it as valid.

As an experiment, I made two alternative HID descriptors. Can you try these commands while SSH'ed into TinyPilot, let it reboot, and then let me know if you have better results?

### Version 1

```bash
sudo su tinypilot bash -c "cd /opt/tinypilot && git pull && git checkout hid-alt1" && \
  sudo reboot
```

### Version 1

```bash
sudo su tinypilot bash -c "cd /opt/tinypilot && git pull && git checkout hid-alt2" && \
  sudo reboot
```

bkortleven, 2020-12-12:
I will test on Monday, when I have access to the equipment.Will provide you the necessary feedback.Thanks

Version 2

bkortleven, 2020-12-14:
hey Michael, 

Tried both options, did a reboot of both the KVM extender the TinyPilot is connected to. 
Updated the TinyPilot software through the update command 
/opt/tinypilot/scripts/upgrade && sudo reboot 

The first option (Version 1 below) gives me keyboard access through the KVM extender. 
Mouse is not working... 

The second option (Version 2 below) gives me no keyboard, nor mouse access... 

So for keyboard access, I'm able to use the TinyPilot with IHSE CON unit? Which is great. Hopefully, mouse control will follow shortly. 

Thanks for the info and resolution! 
Love your product and support! 

Bram

mtlynch, 2020-12-15:
Thanks for confirming! I need to test a bit more to make sure this isn't going to break compatibility with other systems, but I expect to have this fixed in the next release.

ramj9, 2020-12-16:
Having keyboard only work is probably as good as it gets with KVM switches. I have TeSmart KVM which is not recognizing the composite device on main branch, hid-alt1 or hid-alt2. 

This most likely is the reason: [https://superuser.com/a/1558955/364266](https://superuser.com/a/1558955/364266)

Just to verify if thats the case, I am going to disable the usb mouse and see if it works as plain keyboard.

PR #375 (2020-12-23): Update keyboard descriptor HID
Change keyboard HID descriptor to match Microsoft IntelliType keyboard.

Fixes #352
commit 33d3cd56 (2020-12-23): Update keyboard descriptor HID (#375)

Change keyboard HID descriptor to match Microsoft IntelliType keyboard.

Fixes #352
closed 2020-12-23 by mtlynch