tiny-pilot/tinypilot #359: Sending custom keycodes (e.g. Power) from TinyPilot?

victorhooi, 2020-12-06:
I'm looking into whether we can use TinyPilot to remotely control some ChromeOS based devices.

These devices often have specific keys for things like "Search", or "Power".

I can get the keycodes for these - e.g. 0x98 is power.

However, is there any way to send custom keycodes to a device from PiKVM?

And do you know if sending these will still work when the device is powered off?

mtlynch, 2020-12-09:
Thanks for filing this!

That might be possible. I suspect that it won't work when the device is powered off, though.

Can you share screenshots of what appears when you push those keys while viewing [keycode.info](https://keycode.info/)?

mtlynch, 2020-12-31:
Didn't hear back, so I'm closing, but feel free to reopen if you'd like to share screenshots.

closed 2020-12-31 by mtlynch