tiny-pilot/tinypilot #368: HDMI works keyboard does not

majes-git, 2020-12-10:
I have installed tinypilot as described on top of raspios (2020-12-02). See log below. The HDMI part works fine, but keyboard does not. I have tested several USB-C to USB-A cables to no avail.

[image]

What also looks strange is the "red" button indicating "Could not connect to TinyPilot:"
There are also errors in the developer tools see screenshot below. I'm using Chrome.

[image]

Install log:
```
pi@pikvm:~ $ curl -sS https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install \
>   | bash -
+ '[' -z '' ']'
+ TINYPILOT_INSTALL_VARS=
+ echo 'Using default install vars'
Using default install vars
+ USE_TC358743_DEFAULTS=
+ [[ '' =~ ustreamer_capture_device= ]]
+ '[' -f /home/ustreamer/config.yml ']'
+ '[' -z '' ']'
+ [[ ! '' =~ ustreamer_encoder= ]]
+ TINYPILOT_INSTALL_VARS+=' ustreamer_encoder=hw'
+ [[ !  ustreamer_encoder=hw =~ ustreamer_format= ]]
+ TINYPILOT_INSTALL_VARS+=' ustreamer_format=jpeg'
+ [[ !  ustreamer_encoder=hw ustreamer_format=jpeg =~ ustreamer_resolution= ]]
+ TINYPILOT_INSTALL_VARS+=' ustreamer_resolution=1920x1080'
+ [[ !  ustreamer_encoder=hw ustreamer_format=jpeg ustreamer_resolution=1920x1080 =~ ustreamer_persistent= ]]
+ TINYPILOT_INSTALL_VARS+=' ustreamer_persistent=true'
+ [[ !  ustreamer_encoder=hw ustreamer_format=jpeg ustreamer_resolution=1920x1080 ustreamer_persistent=true =~ ustreamer_desired_fps= ]]
+ TINYPILOT_INSTALL_VARS+=' ustreamer_desired_fps=30'
+ set -u
+ set -e
+ echo 'Final install vars:  ustreamer_encoder=hw ustreamer_format=jpeg ustreamer_resolution=1920x1080 ustreamer_persistent=true ustreamer_desired_fps=30'
Final install vars:  ustreamer_encoder=hw ustreamer_format=jpeg ustreamer_resolution=1920x1080 ustreamer_persistent=true ustreamer_desired_fps=30
++ mktemp -d
+ pushd /tmp/tmp.ldxWTRmNdb
/tmp/tmp.ldxWTRmNdb ~
+ sudo apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [346 kB]
Fetched 13.4 MB in 12s (1,123 kB/s)
Reading package lists... Done
+ sudo apt-get install -y libffi-dev libssl-dev python3-dev python3-venv
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dh-python libexpat1-dev libpython3-dev libpython3.7-dev libssl1.1 python-pip-whl python3-distutils python3-lib2to3 python3.7-dev python3.7-venv
Suggested packages:
  libssl-doc
The following NEW packages will be installed:
  dh-python libexpat1-dev libffi-dev libpython3-dev libpython3.7-dev libssl-dev python-pip-whl python3-dev python3-distutils python3-lib2to3 python3-venv python3.7-dev python3.7-venv
The following packages will be upgraded:
  libssl1.1
1 upgraded, 13 newly installed, 0 to remove and 11 not upgraded.
Need to get 52.8 MB of archives.
After this operation, 82.1 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libssl1.1 armhf 1.1.1d-0+deb10u4+rpt1 [1,276 kB]
Get:2 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf python3-lib2to3 all 3.7.3-1 [76.7 kB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf libssl-dev armhf 1.1.1d-0+deb10u4+rpt1 [1,583 kB]
Get:3 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf python3-distutils all 3.7.3-1 [142 kB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf python-pip-whl all 18.1-5+rpt1 [1,591 kB]
Get:5 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf dh-python all 3.20190308 [99.3 kB]
Get:7 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf libexpat1-dev armhf 2.2.6-2+deb10u1 [127 kB]
Get:8 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf libffi-dev armhf 3.2.1-9 [159 kB]
Get:9 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf libpython3.7-dev armhf 3.7.3-2+deb10u2 [47.2 MB]
Get:12 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3-dev armhf 3.7.3-1 [1,264 B]
[... 364 lines cut ...]
changed: [localhost]

RUNNING HANDLER [mtlynch.ustreamer : reload systemd config] ********************
ok: [localhost]

RUNNING HANDLER [mtlynch.ustreamer : restart uStreamer] ************************
changed: [localhost]

RUNNING HANDLER [geerlingguy.nginx : restart nginx] ****************************
changed: [localhost]

RUNNING HANDLER [geerlingguy.nginx : reload nginx] *****************************
changed: [localhost]

RUNNING HANDLER [mtlynch.tinypilot : reload TinyPilot systemd config] **********
ok: [localhost]

RUNNING HANDLER [mtlynch.tinypilot : restart TinyPilot service] ****************
changed: [localhost]

PLAY RECAP *********************************************************************
localhost                  : ok=55   changed=32   unreachable=0    failed=0    skipped=23   rescued=0    ignored=0

pi@pikvm:~ $ sudo reboot
```

mtlynch, 2020-12-10:
Oh, it looks like it's failing to establish a websockets connection. Keystrokes and mouse events go over the websockets channel, so that's why they're failing.

Can you share the logs by running `/opt/tinypilot/dev-scripts/dump-logs`?

majes-git, 2020-12-10:
Here you go http://sprunge.us/Z8NPw9
Please note that I have changed my client from a routed network to the same broadcast domain as the raspberrypi, first the client appeared as the NAT GW (172.29.1.108) later it appears directly (172.29.1.160).

I also run tcpdump on my client to see if there are connection attempts from pikvm, but there are none.

Could this may be related to this error message (from the tinypilot journal)?

```
Dec 10 00:02:27 pikvm python[614]: The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)
Dec 10 00:02:27 pikvm python[614]: [2020-12-10 00:02:27,930] ERROR in server: The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)
```

majes-git, 2020-12-10:
I have also tried Firefox and Safari, as well as another laptop. Same behavior.

majes-git, 2020-12-10:
My browser gets this as well as a response by nginx to all the /socket.io/ requests (which return a 400):

```
"The client is using an unsupported version of the Socket.IO or Engine.IO protocols"
```

majes-git, 2020-12-10:
Looks like the socket.io/engine.io version in nginx is too old. After downgrading the pip packages it somehow seems to work:

```
$ sudo -u tinypilot /opt/tinypilot/venv/bin/pip install 'python-socketio==4.*'
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-socketio==4.*
  Downloading https://files.pythonhosted.org/packages/b2/91/d78afbe5861de53cafe9dcc6eb1a802d5495cfbffb79b8385ee7cf15a71c/python_socketio-4.6.1-py2.py3-none-any.whl (51kB)
    100% |████████████████████████████████| 61kB 821kB/s
Requirement already satisfied: six>=1.9.0 in ./venv/lib/python3.7/site-packages (from python-socketio==4.*) (1.15.0)
Collecting python-engineio<4,>=3.13.0 (from python-socketio==4.*)
  Downloading https://files.pythonhosted.org/packages/8d/35/b9edf38b6a42077f3c097ab4edb0281adb3cc5434645c0bd301b0b45c767/python_engineio-3.14.2-py2.py3-none-any.whl (51kB)
    100% |████████████████████████████████| 61kB 2.4MB/s
Installing collected packages: python-engineio, python-socketio
  Found existing installation: python-engineio 4.0.0
    Uninstalling python-engineio-4.0.0:
      Successfully uninstalled python-engineio-4.0.0
  Found existing installation: python-socketio 5.0.1
    Uninstalling python-socketio-5.0.1:
      Successfully uninstalled python-socketio-5.0.1
Successfully installed python-engineio-3.14.2 python-socketio-4.6.1
```

However there are still 2 major issues:

1. The tinypilot journal is being flooded by messages as such:
```
Dec 10 09:29:58 pikvm python[618]: Process Process-1:
Dec 10 09:29:58 pikvm python[618]: Traceback (most recent call last):
Dec 10 09:29:58 pikvm python[618]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Dec 10 09:29:58 pikvm python[618]:     self.run()
Dec 10 09:29:58 pikvm python[618]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Dec 10 09:29:58 pikvm python[618]:     self._target(*self._args, **self._kwargs)
Dec 10 09:29:58 pikvm python[618]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
Dec 10 09:29:58 pikvm python[618]:     hid_handle.write(bytearray(buffer))
Dec 10 09:29:58 pikvm python[618]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Dec 10 09:29:58 pikvm python[618]: Process Process-2:
Dec 10 09:29:58 pikvm python[618]: Traceback (most recent call last):
Dec 10 09:29:58 pikvm python[618]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Dec 10 09:29:58 pikvm python[618]:     self.run()
Dec 10 09:29:58 pikvm python[618]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Dec 10 09:29:58 pikvm python[618]:     self._target(*self._args, **self._kwargs)
Dec 10 09:29:58 pikvm python[618]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
Dec 10 09:29:58 pikvm python[618]:     hid_handle.write(bytearray(buffer))
Dec 10 09:29:58 pikvm python[618]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Dec 10 09:29:59 pikvm python[618]: Process Process-3:
Dec 10 09:29:59 pikvm python[618]: Traceback (most recent call last):
Dec 10 09:29:59 pikvm python[618]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Dec 10 09:29:59 pikvm python[618]:     self.run()
Dec 10 09:29:59 pikvm python[618]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Dec 10 09:29:59 pikvm python[618]:     self._target(*self._args, **self._kwargs)
Dec 10 09:29:59 pikvm python[618]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
Dec 10 09:29:59 pikvm python[618]:     hid_handle.write(bytearray(buffer))
Dec 10 09:29:59 pikvm python[618]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
```
2. The latency of the screen/HDMI is way too high - it is several seconds (2secs up to 10secs). Based on your post it should be around 200ms. I think I have the same model of the video capture device (`534d:2109`). At least `v4l2-ctl` shows `Motion-JPEG` support.

Any ideas what else I can do to fix the keyboard issues and to improve the HDMI latency?

majes-git, 2020-12-10:
That's strange. When I connected the raspberrypi to ethernet instead of wifi the error messages disappeared and the latency improved tremendously. The client is still on wifi and normally I don't have issues with the wireless connections (using Unifi equipment).

Is it possible that the wifi chip in Raspi 4 is just crap? Any suggestions how to operate the tinypilot over wireless networks?

mtlynch, 2020-12-10:
Thanks for the heads up! This turned out to be [a bug in Flask-SocketIO](https://github.com/miguelgrinberg/python-socketio/issues/578#issuecomment-742477016). I fixed it in #370 

You should be seeing performance a little bit slower over WiFi but not 10-50x slower.

Are you still seeing those errors in the logs?

majes-git, 2020-12-11:
Thanks Michael!

The fix works well. Over LAN I don't see those error messages. Wifi is still slow like hell it it seems to delay is stacking up. I'm going to get a USB wifi device and see if that has better performance.

closed 2020-12-16 by majes-git