tiny-pilot/tinypilot #373: Option for video pass-through

champignoom, 2020-12-12:
Currently it goes like
```
server <==usb(hid)==> tinypilot <==ip(hid+video)==> localhost
  |--------video-------->
```

But when the server and the localhost are geographically close, it should be possible for the video to bypass tinypilot:
```
server <==usb(hid)==> tinypilot <==ip(hid)==> localhost
   |--------------------video------------------>
```

This way, the whole workload of video compression and transmission is freed from tinypilot. Pi Zero W should then be a feasible alternative device (see https://github.com/mtlynch/tinypilot/issues/93).

This could be implemented by a http interface for usb on localhost + a video source picker in the UI.

mtlynch, 2020-12-12:
Thanks for filing and for the suggestion!

This unfortunately is not possible. For the server to send video directly to the user's browser, there would need to be a TinyPilot agent application running on the server to send the video data. But if you can run software on the target server, then TinyPilot likely isn't the right tool, because the value of TinyPilot is that you don't need special software running on the target device.

Put another way, imagine that you're trying to capture a BIOS interface in TinyPilot. I'm not aware of any BIOS that allows you to redirect video to a remote websocket daemon, so the direct connection solution wouldn't work.

closed 2020-12-12 by mtlynch

champignoom, 2020-12-12:
Thanks for the quick response.

But I'm not sure I understand your points, because all it takes to let the server send video data is a single hdmi cable plugged into it (with a hdmi capture dongle on the other side). No TinyPilot agent should be necessary.

mtlynch, 2020-12-13:
That's an interesting alternative, but then the user's computer would need extra software on it to process the incoming video. I'd like to keep TinyPilot's requirements on the end-user machine to browser-only with no additional software.

champignoom, 2020-12-13:
Indeed there would be hassle on the user's side, therefore the status quo should definitely remain as the default.

But when the user experiences more lag than they can bear, it would be nice if there's an option to reduce the lag with a one-off cost. Ideally the user only has to run a one-liner copied from the UI page, something like `curl http://raspberrypi.local/dongle_deamon_linux.sh | sh`, so the workflow is not drastically interrupted.

mtlynch, 2020-12-14:
Unfortunately, it would add too much complexity for too niche a scenario. We'd have to provide installers for each supported client OS, which is a substantial amount of work.