tiny-pilot/tinypilot #386: quick-install  for error

TelDragon, 2020-12-20:
run quick-install
```
TASK [geerlingguy.nginx : Ensure nginx service is running as configured.] *************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Service is in unknown state", "status": {}}
```
chenk nginx status ....

```

systemctl status nginx
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-12-20 09:43:42 UTC; 3min 1s ago
     Docs: man:nginx(8)
 Main PID: 5888 (nginx)
    Tasks: 5 (limit: 906)
   Memory: 5.3M
   CGroup: /system.slice/nginx.service
           ├─5888 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
           ├─5889 nginx: worker process
           ├─5890 nginx: worker process
           ├─5891 nginx: worker process
           └─5892 nginx: worker process

Dec 20 09:43:42 orangepizero systemd[1]: Starting A high performance web server and a reverse proxy server...
Dec 20 09:43:42 orangepizero systemd[1]: Started A high performance web server and a reverse proxy server.
```

TelDragon, 2020-12-20:
I did this.

/opt/tinypilot-updater/geerlingguy.nginx/tasks/main.yml

```
 #- name: Ensure nginx service is running as configured.
      #  service:
      #    name: nginx
      #    state: "{{ nginx_service_state }}"
      #    enabled: "{{ nginx_service_enabled }}"
```

TelDragon, 2020-12-20:
```
TASK [mtlynch.tinypilot : create TinyPilot virtualenv] ********************************************************************************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ModuleNotFoundError: No module named 'pkg_resources'
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to import the required Python library (setuptools) on orangepizero's Python /usr/bin/python3. Please read module documentation and install in the appropriate location. If the required library is installed, but Ansible is using the wrong Python interpreter, please consult the documentation on ansible_python_interpreter"}
```

chenk
```
root@orangepizero:~# . /opt/tinypilot-updater/venv/bin/activate
(venv) root@orangepizero:~# pip install pkg_resources
Requirement already satisfied: pkg_resources in /opt/tinypilot-updater/venv/lib/python3.7/site-packages (0.0.0)
(venv) root@orangepizero:~# pip install setuptools
Requirement already satisfied: setuptools in /opt/tinypilot-updater/venv/lib/python3.7/site-packages (40.8.0)
```

TelDragon, 2020-12-20:
/opt/tinypilot-updater/mtlynch.tinypilot/tasks/main.yml
```
- name: save absolute path to python3
  set_fact:
    python3_abs_path: "{{ realpath_python3.stdout }}"

    #- name: create TinyPilot virtualenv
    #  pip:
    #    virtualenv: "{{ tinypilot_dir }}/venv"
    #    virtualenv_command: "{{ python3_abs_path }} -m venv venv"
    #    requirements: "{{ tinypilot_dir }}/requirements.txt"
  notify:
    - restart TinyPilot service
```

TelDragon, 2020-12-20:
What puzzles me is whether the ustreamer service lacks parameters? Should I edit it myself? E.g. /dev/video1

--device=/dev/video1

/lib/systemd/system/ustreamer.service

```
[Service]
Type=simple
User=ustreamer
WorkingDirectory=/opt/ustreamer
ExecStart=/opt/ustreamer/ustreamer \
  --host 127.0.0.1 \
  --port 8001 \
  --encoder hw \
  --format jpeg \
  --desired-fps 30 \
  --resolution 1920x1080 \
  --persistent \

Restart=always

[Install]
WantedBy=multi-user.target
```

Or are there selectable items on the WEB page?

mtlynch, 2020-12-21:
As I [mentioned](https://github.com/mtlynch/tinypilot/issues/376#issuecomment-746987153) on #376, Raspberry Pi is the only system that TinyPilot supports at the moment.

You've now filed four separate bugs in five days about not being able to install on an orangepizero. Please stop filing bugs about this issue.

closed 2020-12-21 by mtlynch